cleanflight/cleanflight #1905: no conection to cleanflight ?

olaf2273, 2016-02-29:
hello I have Problems whit flashing cleanflight:

1* Naze32 bootpins shortened ,(reboot)
Wen: button on in cleanflight.
No reboot sequens
Flash on connect 
thenLoad firmware, then comet the naze32. = Bleu light on - no flickering.
than response from cleanflight !!!

2016-02-29 @ 17:16:53 -- Running - OS: Windows, Chrome: 48.0.2564.116, Configurator: 1.2.1
2016-02-29 @ 17:18:23 -- Detected: COM3 - triggering flash on connect.[/color]

Noting happened !!!, when enter flash.
than response from cleanflight !!!
no response from bootloader-programing failer !

2* naze32 no boot pins shortened: 
Wen naze32 connecting to computer, Lights bleu/orange all flickering
open clean flight:
2016-02-29 @ 17:15:46 -- Running - OS: Windows, Chrome: 48.0.2564.116, Configurator: 1.2.12016-02-29 @ 17:16:06 -- Serial port successfully opened with ID: 8
2016-02-29 @ 17:09 -- Failed to open serial port
2016-02-29 @ 17:16:16 -- No configuration received within 10 seconds, communication failed
no response from bootloader-programing failer !

I download CP210x drivers.
I did re COM3  with : hercules_3-2-8.exe
and i did  STMFlashLoaderDemo
but with STMFlashLoaderDemo,
also message:
No response from the target, the boot Loader can not be stared
Please, verify the boot mode configuration and flash protection status.
Reset your device then try again ..

can some one help me ?
thanks 
greetings Olaf

AmiSMB, 2016-03-01:
I would not use the boot pins as I like to unplug and plug back in cycling the power. I think you should open device manager in windows and look under Ports (COM & LPT) section. What you should see is that when you connect your board to the USB port it will update the Ports list and your new port should appear and from this you will be able to see which COM port it is installing as. I think if the Clearflight auto connect is ticked then the software should automatically detect the COM port and board as you connect it. You may find this video helpful https://www.youtube.com/watch?v=nVF-gQsUu50

joshuabardwell, 2016-03-03:
This is a support question. Github is for bug tracking and development requests. Please close this issue. You can get support help on the RCGroups Cleanflight thread.

http://www.rcgroups.com/forums/showthread.php?t=2249574

closed 2016-03-09 by hydra