cleanflight/cleanflight #1911: Tricopter motor mix broken in master

lkaino, 2016-03-01:
I was rebasing my fork of cleanflight to the latest master. After the rebase the motor mix wasn't working correctly for tricopter anymore. Motor 1 controlled actually motor 3 and other motor outputs controlled nothing.

I was searching for issue in my merges but finally noticed that it is this commit that broke it in master: [f54fa52](https://github.com/cleanflight/cleanflight/commit/f54fa52c129076dd13d171decf02d5105a1a5cb0).

Yes, I should rebase to the tested releases.

ledvinap, 2016-03-01:
What target so you use?

lkaino, 2016-03-01:
Ah forgot to mention, target is naze. Seems like duplicate.

ledvinap, 2016-03-01:
here is the problem:
https://github.com/cleanflight/cleanflight/commit/f54fa52c129076dd13d171decf02d5105a1a5cb0#diff-0eb424e2144286476c50709ab5aada87R626

enum is used in `#if` condition ...

ledvinap, 2016-03-01:
`-Wundef` will warn about this. And reveal another problem:
```
./src/main/drivers/pwm_rx.c:44:25: warning: "MAX_PWM_INPUT_PORTS" is not defined [-Wundef]
 #if PPM_CAPTURE_COUNT > MAX_PWM_INPUT_PORTS
```

lkaino, 2016-03-05:
Fixed in referenced case.

closed 2016-03-05 by lkaino