cleanflight/cleanflight #1917: EMF avoidance messes up motor timing

borisbstyle, 2016-03-02:
Just to report. EMF avoidance enabled sends wrong motor pulse lengths to the ESC. This is because of the 7/6th factor on the timing difference.

Solution/workaround is to apply the the correction on the motor value.

ledvinap, 2016-03-02:
Timer prescaler can be configured according to emf avoidance. Counter frequency will stay the same,  minimizing code affected .

borisbstyle, 2016-03-03:
yeah prescaler is what works well. Just do value * 7 /6 and you will get the correct timing

ledvinap, 2016-03-03:
It is strange -  `SystemCoreClock` is used when initializing timer. This value is probably wrong..

What does cli `status` show?

borisbstyle, 2016-03-03:
84Mhz.

It seems to only happen with oneshot125. Try switching back and fourth between emf-avoidance and now emf_avoidance and you will see that your motor pulse will shift. The props would not spin until you readjust the min_throttle.
Adding a prescaler of 7/6 fixes the issue and motor spin at same value

ledvinap, 2016-03-03:
One problem is that oneshot frequency is 8MHz, but 84 is not divisible by 8 (84 / 8 = 10.5), so resulting timer frequency is 8.4MHz.

Using 7/6 compensation should not fix this ...

ws2811 is affected too (84/24 = 3.5), using 12MHz instead of 24MHz should work fine (all trigger are divisible by 2)

borisbstyle, 2016-03-03:
@ledvinap 
On my branch I use 24mhz timer for oneshot. I do this so I can get oneshot42 support too. Even there problem exists with EMF avoidance and 7/6 compensation really fixed it.

ledvinap, 2016-03-03:
Yes, the difference is 7/6 ith 24MHz. But the compensation won't work with 8MHz .. 

Actual timer frequency may be used for pulse length calculation (instead of desired value that is used now). It may even simplify timer sharing etc.

hydra, 2016-03-09:
@borisbstyle can you create a PR for the oneshot42 support and include this fix?  Seems like a good approach.

@ledvinap If you wait for boris to add the oneshot42 support then maybe do a PR to use timer frequencies that would really be nice.

borisbstyle, 2016-03-09:
I will make a PR with oneshot42 support and also change the current oneshot
timer to scalable value. I will use 24mhz for both

closed 2017-02-25 by hydra