cleanflight/cleanflight #1921: OneShot 125 support on 1.12 with EMAX BLHeli 20A

lainchir, 2016-03-04:
Hardware - Naze Rev 6 - EMAX BLheli 20A (Oneshot enabled) - EMAX 2216 810kv

Just flashed 1.12, and as soon as I enable OneShot, and reboot, my motors go into slow speed spin. Nothing will turn them off again. As soon as I disable it, they are fine again. OneShot works fine for me in 1.11

3djc, 2016-03-04:
Try disabling gyro_sync, or try gyro_sync_denom set to 2

3djc

lainchir, 2016-03-04:
Isn't gyro_sync disabled by default in 1.12? I will check as soon as I have time to check.

Will give gyro_sync_denom at 2.

Currently rolled back, but will try soon as I get 5. 

Many thanks :)

dawsoncc, 2016-03-05:
I'm having the same problem here. :-(
Also using emax BLheli20A and naze

lainchir, 2016-03-05:
Hi 3DJC.

Issue resolved with CLI command of "set gyro_sync = OFF"

I'm sure in the Rel 1.12 manual it states that this is disabled by default... pg 7? Wondering therefore if this has been released with it enabled in error?

Many thanks for your help

Ian

closed 2016-03-05 by lainchir

borisbstyle, 2016-03-05:
It is not disabled by default. The issue is blheli firmware pre 14.2 (14.1 or older). Not the cleanflight firmware

lainchir, 2016-03-05:
Hi Borisbstyle.

Thanks for the info. I did wonder about the BLHeli versions on the eMax, but trying to update then through cleanflight and the BLHeli flasher just shows 4 unknown esc's. I guess I need to do them manually with an arduino board or the like? Will it make a huge difference to have the gyro_sync off to a noob flying a 450 camera platform? (I suspect not)

Thanks for all your work with these boards :)

borisbstyle, 2016-03-05:
well even without gyro sync it might take longer to arm sometimes as the
older blheli versions can struggle with 1000looptime.

Those escs dont have blheli bootloader and thats why the update doesnt work
through cleanflight :(

lainchir, 2016-03-05:
Quick trawl of the internet shows that in a bit more detail. It's above my head for sure! :)

I'm now at 1.12 with gyro_sync off, so all good.  Thanks for the explanation Borsibstyle.

Many thanks

3djc, 2016-03-06:
I'll have a look at the doc you mention. Glad to hear it is solved