cleanflight/cleanflight #1925: Unable to set airmode in CLI

kgrammatis, 2016-03-05:
According to the ['Modes' documentation](https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md) AirMode is available to use by manually assigning it to an AUX channel in the CLI. 

I used the following command in the CLI but there was no effect.
aux 4 27 4 1400 2100

The results:
# aux
aux 0 0 0 1300 2100
aux 1 2 1 1300 2100
aux 2 25 0 1600 2100
aux 3 20 3 1300 2100
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

#

hydra, 2016-03-09:
I tested this, worked fine.
```
# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# aux 4 27 4 1400 2100

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 27 4 1400 2100
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

save
```
[image]

I will update the documentation now if needed.

hydra, 2016-03-09:
i couldn't spot any of the mode ids being wrong.  make sure you look at the right column

closed 2016-03-09 by hydra

hydra, 2016-03-09:
Modes documentation updated in d96123d8f78309018542c834f4ade11716ec8b55

Thanks for the report.