cleanflight/cleanflight #1928: Looptime does not affect cycletime

robjahnke, 2016-03-06:
Changing the looptime does not seem to affect cycle time on NAZE32. I have set looptime=1500 and also  set looptime=2500 with a save each time. The cycle time continues to report 990 - 1005 microseconds in the GUI. This has been verified with a Blackbox recording as well. It is acting as though looptime = 0 which would mean "run as fast as you can".

# version
# Cleanflight/NAZE 1.12.0 Feb 19 2016 / 20:01:21 (aa8608a)
# 
[image]

[image]

Thanks, robjahnke

borisbstyle, 2016-03-06:
read manuals......looptime doesnt work when gyro_sync is enabled

robjahnke, 2016-03-06:
Thanks. I've been reading the manuals a lot, but I missed that point. Issue Closed :-)  Rob

closed 2016-03-06 by robjahnke