cleanflight/cleanflight #1932: throttle too low, won't arm

proggod, 2016-03-06:
Throttle was showing 885 in configurator, so the quad would never arm.

proggod, 2016-03-06:
Range logic should be anything below a minimum number should allow the arm switch to arm the motors, there is some arbitrary range that is used to allow arming.  Is there a reason its not just under a certain number?  If no on agrees with my logic, then remove this ticket.  Yes I know how to fix it with rxrange command, and no idea why all of a sudden radio switched to a lower number.  But I don't understand the logic here..

proggod, 2016-03-06:
Line 115 of rc_controls.c looks like logic works like I said, any idea WHY it wouldn't arm?  Am I missing something?

HaukeRa, 2016-03-06:
It's probably failsafe.

proggod, 2016-03-06:
http://prntscr.com/ac3sy0
http://prntscr.com/ac3t5f
wont arm right now, why?

proggod, 2016-03-06:
It is failsafe causing problem, i think its because switches are at 999, this is after rxrange tuning on all of them, there needs to be some tolerance put in on these values, it can easily be +-5

proggod, 2016-03-06:
Although flicking arm back and forth a few times, it eventually will arm.

proggod, 2016-03-06:
Also in config for failsafe valid range is 885-2115 so failsafe should not be triggering

tozes, 2016-03-07:
Same problem here, sometimes I have to switch back and forth 10+ times until it arms.

hydra, 2016-03-09:
see #1895

closed 2016-03-09 by hydra

proggod, 2016-03-09:
Does this fix non-GPS problems ?    I have this arming problem on 5 different quads, w/ multiple boards, CC3D/Cleanflight, Sparky/Cleanflight, Revolution/Raceflight.   Looks to me like that changes the loop time, which shouldn't be an issue on those other boards.

tozes, 2016-03-10:
You could always try it and see if it helps. In my case I think the F1 on the naze32 couldn't cope with all the features and maintain 1000 looptime.

You're right, changing `gyro_sync_denom = 2` changes the looptime to 2000. I tried to set 1.5 but it didn't work, the command seems to accept only integers.

borisbstyle, 2016-03-10:
You cannot use denom of 1.5

gyro_sync_denom is the amount of gyro samples you want to have compared to
the max amount of samples.

denom 1 means 1:1 samples
denom 2 means 1:2 samples
denom 3 means 1:3 samples

you cant get 1.5 sample

proggod, 2016-03-10:
So your telling me my sparkys and revolutions with nothing activated other then horizon mode can't handle the default loop times?   Even if it is the loop time, shouldn't it eventually see switch is armed if you leave it on?  Sometimes it never even arms, but by the time I plug it into a PC, it arms..

borisbstyle, 2016-03-10:
than there is somethint else on yours....check basic stuff like endpoints
min_check etc.

It works on all other targets just fine

proggod, 2016-03-10:
Endpoints are all between 999 and 2001, this is with multiple versions as, raceflight & cleanflight, and multiple build versions, and I have talked to others with same issues.   I am not sure why this ticket was closed..

proggod, 2016-03-10:
Will black box logs show what is going on?

borisbstyle, 2016-03-10:
It sounds like a config issue to me.

What board are you using?

borisbstyle, 2016-03-10:
check receiver tab what its doing....check leds on the board....check cpu
usuage with status command. Check if your sticks are nicely centered.

Also try arming on a switvh as well

proggod, 2016-03-10:
On raceflight its an Revolution, in cleanflight they are sparkys and cc3ds.  The only thing in common is we always use spektrum radios.   But I personally witnessed it not arming in the config.  Look at my screenshots, everything is perfect, the throttle was perfect, and yet it didn't glow green in cleanflight when arm switch was depressed.   You can see in the configurator the arm switch was in the right position too.

This is with switch arming, refer to the screen shots at the top.

hydra, 2016-03-10:
Blackbox logging will not help as logs are not made when disarmed.

Post a config dump please.

proggod, 2016-03-10:
[cc3d_dump.txt](https://github.com/cleanflight/cleanflight/files/167888/cc3d_dump.txt)
[Sparky_dump.txt](https://github.com/cleanflight/cleanflight/files/167887/Sparky_dump.txt)
[CC3D_revo_dump.txt](https://github.com/cleanflight/cleanflight/files/167889/CC3D_revo_dump.txt)

Here you go, 3 different boards all seem to have same issue.

CanEHdian101, 2017-05-09:
i am having a similar issue but the led is showing green on my name 32 but in the config it doesn't light up

mikeller, 2017-05-09:
@CanEHdian101: Check CPU load (should be well below 100%).

CanEHdian101, 2017-05-10:
@mikeller it it at like 7%

mikeller, 2017-05-10:
Ok, that's not it then. Some more things to check: Throttle at 0, gyro init finished (3 beeps).

CanEHdian101, 2017-05-10:
So get no beeps what so ever unless I unplug receiver they plug in the battery and reconnect the receiver. I know but it's weird but it tells me all the hardware works

mikeller, 2017-05-10:
That's a weird one. Can you post a config `diff` please?

CanEHdian101, 2017-05-11:
I am kinda new so I don't know what a config  diff is so could you please tell me and now after a re flash it shows in the config but still no tones at all

mikeller, 2017-05-11:
You should probably read up here then: https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21