cleanflight/cleanflight #1933: v1.12: S.Port Telemetry via softserial not working anymore (Naze32 Afro rev3)

Murphy-c, 2016-03-06:
Issue: FrSky S.Port Telemetry doesn't work on CF v1.12, when using Softerial1.

How to reproduce:
-hardware: FrSky x4r-sb, Naze Afromini rev3
 put diode on RC6 to RC5, connect that to x4r-sb S.port
-flash 1.12 with full chip erase
 Config:
  feature -RX_PARALLEL_PWM
  feature VBAT
  feature RX_SERIAL
  feature SERVO_TILT
  feature SOFTSERIAL
  feature TELEMETRY
  feature ONESHOT125
  set serialrx_provider = SBUS
  serial 1 64 115200 57600 0 115200
  set telemetry_inversion = on
  gyro_sync_denom = 2
-- save & reboot then add --
  serial 30 32 115200 57600 0 115200

The same setup works on v1.11 (minus the gyro_sync_denom = 2 which is only necessary on v1.12 as it seems to be running out of cycles on F1 targets)

Things I tried to debug without success:
- gyro_sync = off
- looptime = 3500
- forcing softserial telemetry to 57600 (is in auto by default)
-telemetry_inversion = off

hydra, 2016-03-09:
can someone else confirm this please.  Current softserial is only good for low-speed devices up to 19200 as such this is pretty low priority at the moment.

If another developer has time could they please investigate?

3djc, 2016-03-09:
I'll try to duplicate

ledvinap, 2016-03-09:
SPort is 57600.. It was IMO never officially supported..

3djc, 2016-03-09:
I could not reproduce with gyro sync off
I could not reproduce with gyro sync on, denom = 2

The only difference I can think off is I'm not using a diode but a simple short between RC5-RC6

Config : http://pastebin.com/NPVdJ4ck

Murphy-c, 2016-03-13:
I've tried 1.12 with 3djc's dump and still get no telemetry. I tried removing the diode and now get no telemetry even on 1.11, so it could be that the tolerance of the batch of IC used on my boards are a lower than on 3djc's.
I also tried to change the power source to a more stable 5v, still not better.

I understand the Low priority label and hope we can get someone else to try and reproduce it in the future, just to make sure if it's a hardware issue on my side or not.

3djc, 2016-03-13:
I think what this means is that we cannot reliably say it is working, nor will. It may in some case, not in others. Softserial is probably not a great idea given the speed required for smartport, maybe it would be more sensible to find a board that could do this in hardware

ghostsharp, 2016-04-08:
Murphy, in my case, i got that work, without diode and using the same baud rate the S.Port and GPS Port. I used 57600 for both. (Naze32 and X8R (mode D16) and 9XR Pro.(mode D16), set telemetry_inversion = ON)

Ede2016, 2016-04-15:
I had telemetry working on SoftSerial with the AfroMini (rev3, the current version) together with the FrSky XSR or with the X4R-SB. I used 10k resistors from the TX and the RX and then connected the resistors to the SmartPort.
Here is what I did: http://fpvlab.com/forums/showthread.php?46100-New-FrSky-XSR-–-3-8g-–-buy-it-now&p=781540&viewfull=1#post781540
This worked with CleanFlight 1.11.0. I could set the baud rate to Auto or anything. It always worked.
Now I upgraded to CleanFlight 1.12.1 and telemetry on my Taranis only discovers the receiver telemetry data like RSSI but not anymore the FC telemetry data.
I never used CleanFlight 1.12.0 so I don't know if that works.

Ede2016, 2016-04-15:
@hydra: I am a software developer (Windows, VB, C#) and I like to contribute. For a start I would like to compare the code from version 1.11.0 with the code from version 1.12.1 to check why the above described problem happens. Unfortunately I have no experience with GitHub or any other open source programming. I would like to start with downloading the code from version 1.11.0 and 1.12.1 to compare them. But unfortunately I don't find the code for any earlier version. Where is it? Does GitHub have a feature to compare two code versions to check for differences? Please let me know how I can start with this.

3djc, 2016-04-15:
The "problem" is known I think, softserial is not a good option for 57k transmission. It might work in some cases, not in others. Computing load as increased between 1.11 and 1.12, here is your issue. Best solution is to use hardware ports for smartport.

Ede2016, 2016-04-15:
@3djc Thanks for your update. Currently I can't change my hardware because I use a AfroMini in a micro quad and everything else is too big. Is there any hope that in the future SmartPort will work again on SoftSerial or is this something which is in principle difficult to impossible with the Naze32/AfroMini processor so basically it will never work reliable without hardware update.

3djc, 2016-04-16:
What you have to understand, is that it CAN work, but there is no garanty it will, it really depends on your setup and settings. The more you ask to the FC, the less likely it is to keep working. The issue is that there is a too wide range of setup and settings to say for sure when it will, and when it wont work. Things like gyro_sync, low looptimes, heavily loaded ports activity, all of those things are good candidates to break it. Moving forward, I believe Hydra is thinking of some kind of 'on demand' firmware. By removing things you do not need, it might free some ressource to have it work, but the only reliable, garanted way of having it work is to have a controler/setup where you can assign it to a hardware UART

Ede2016, 2016-04-16:
@3djc Thanks for the additional information - understood.
After some more testing I have to admit my problem is clearly not the CleanFlight update. Today I installed version 1.11.0 again and SPort Telemetry still did not work. After some more testing I found out that the SPort of my XSR receiver does not work anymore. I used a X4R receiver with CleanFlight 1.12.1 and telemetry works again. Sorry for all the hassle, CleanFlight is not to blame, it was clearly my error.

martinbudden, 2016-04-28:
@Ede2016 , can you close this issue please?

Ede2016, 2016-04-28:
Close
It seems Murphy-c and I had both other issues which caused this problem.

Ede2016, 2016-04-28:
Close

Ede2016, 2016-04-28:
@martinbudden  
I like to close it but I don't know how. I am not sure I am allowed to close this because I did not open this issue.

cyril-grosset, 2016-10-13:
It appears to me that connecting smartport to a soft serial port works randomly. I can power-up my sp f3 board ten time in a row, and it will works one or two times out of the ten attempts. I know that the best answer regarding those issue is that 'soft serial is slow, and probably not the best choice in conjunction with smartport, and that it might sometimes work and sometimes not. But I looked at smartport.c and I start wondering why some lines of code exists there.

`if (lastChar == FSSP_START_STOP) {
        smartPortState = SPSTATE_WORKING;
        if (c == FSSP_SENSOR_ID1 && (serialRxBytesWaiting(smartPortSerialPort) == 0)) {
            smartPortLastRequestTime = now;
            smartPortHasRequest = 1;
            // we only responde to these IDs
            // the X4R-SB does send other IDs, we ignore them, but take note of the time
        }
    }`

I.e. why is the last request time recorded only for requests cleanflight handle ? Any request should count, when it comes to detect the inactivity.

Also, what is the exact purpose of this timeout ?

`    if ((now - smartPortLastRequestTime) > SMARTPORT_NOT_CONNECTED_TIMEOUT_MS) {
        smartPortState = SPSTATE_TIMEDOUT;
        return;
    } `

SMARTPORT_NOT_CONNECTED_TIMEOUT_MS is set to 7000

So if I read this correctly, if for any reason even in the middle of a flight, the telemetry link is lost for like 7 seconds ( at least regarding the monitored ID ), then the smartport feature will be turned off for the rest of the flight (what about my antenna tracker !?) ? Should you not at least limit this check to the MSP port, and completely ignore it for any other UART ?

Finally this can kill the whole thing too easly:

`        // Ensure we won't get stuck in the loop if there happens to be nothing available to send in a timely manner - dump the slot if we loop in there for too long.
        if ((millis() - smartPortLastServiceTime) > SMARTPORT_SERVICE_TIMEOUT_MS) {
            smartPortHasRequest = 0;
            return;
        }`

SMARTPORT_SERVICE_TIMEOUT_MS is set to 1
I know one millis is an eternity, but still, look at the code executed before the begging of the enclosing loop, if for any reason it takes more than 1 millis, then again everything will stop working.
From my point of view the smartPortLastServiceTime variable should be reset to millis() just before entering the while loop.

Regards 
Cyril

cyril-grosset, 2016-10-13:
I should add one or two things.
First of all I'm far from being a good c/c++ programmer, but I'm a very experienced 40 years old software engineer. I started programming at 12 in Basic on my CPC464, I later switched to Pascal and then Delphi. Finally, I learned java and I'm now working as a senior java developper for many years. I understand low level very well, and for instance I designed an MCU from scratch for fun using the uni-hamburg.de Hades simulator. So although I'm not the best C/C++ dev around, I pretty sure I can help if the Cleanflight team is ready to support my efforts. So I could fork the project, and submit a pull-request, but I have no time to waste. So I would like some feedback from the team here before I spend more time analysing this.

hydra, 2016-10-13:
don't use softserial for smartport.  for LOW SPEED applications ONLY. <= 19200 baud.  Loading the CPU will make it even worse.

@cyril-grosset IIRC the delays are delays from BOOTUP time.  If the smartport receiver is not connected it will not be used even if later connected.  if you have any fixes please check the code, try your fixes, test the changes appropriately using your equipment and possible also with protocol analyzers and then make a PR with your fixes and findings.

closed 2016-10-13 by hydra

cyril-grosset, 2016-10-21:
I spent some more time on this issue, because, when I first power-up the system it works usually so well, and using softserial for this purpose would allow peoples to save a precious UART port.

Additionally although the smartport baudrate is too high for softserial, the smartport protocol itself is not quite 'verbose', since a request is sent every 11ms and basically consist of two bytes, the start/stop byte, and the 8bit physical sensor id. The response, when present, is height bytes long. So in the worst case scenario, (1000ms / 11ms) * (2+8) bytes, the transfert rate on the smartport bus will be around 900 bytes / s.  Also this protocol is very resilient, because the same data are repeated again and again, and if some requests / responses are lost, it doesn't really matter as long as it succeed frequently enough.

With all this in mind, I looked at the code again, and at that point I realised that the reason behind the issue with smartport and softserial, was that because of the inefficiency of softserial, the transmit and receive buffer of the softserial port would gently fill up, causing a lot of latency between the time a request is receive, and the time the response is sent. That much, that the response would be emitted to late (remember the 11ms time frame). 

So I tried the following approach, I added two functions allowing me to flush the content of the transmit and receive buffer for a given serial port. Then I ensured that the receive buffer always contains at most 2 bytes, and that the transmit buffer is cleared before sending a new response. 

And tada, suddenly, I got a totally stable smartport connection between my FC and my x9d, with a very high and constant refresh rate.

Now there is one thing I didn't test, and to be honest that I don't really now how to test, it's to increase the CPU load to check whether it still works correctly under load or not. But still the result is very encouraging because it's basically night and day under the current test conditions.

Also the same result could be achieved in different way, I'm actively flushing the buffers while we could simply resize them when used with smartport, and reset their size for other usages.

I will provide a pull request pretty soon, but in the mean time, if you guys have any thought regarding this, or any suggestion how to increase the CPU load (ok I could just abuse it), you are very welcome.

Regards
Cyril

acc-, 2016-12-30:
@cyril-grosset it looks you didn't provide a pull request, could you please share your change? I'd like to test it on my own.

Daniel-1276, 2018-04-16:
@cyril-grosset looks like you found a smart solution stabilizing smart port transmission over soft serial. Would you mind sharing your code modifications?

cyril-grosset, 2018-04-16:
Oh my, I completely forgot about this. Mainly because right after this I ordered a DJI Mavic, so I forgot about my needs for additional serial ports (since the main requirement for me was to handle a GPS for return to home purposes). And then I just used my FPV drone for short range flights, were return to home was not needed, and my Mavic for long range ones. But also because right after posting this, I realised than, everything was fine with my change as long as I performed the tests outside, by pretty low temps. As soon as I started to experiment at ambient temperature, I realised that the issue was back. I was on my way to investigate some more, but at this point, I encountered a lot of issue in my personnal life (that I won't disclose here but) that caused me to completely leave this asside. I will check whether the code changes still exist somewhere, and eventually submit a pull request just to close the loop. But honnestly I'm not very confident it will solve the issue in any conditions. Also considering the new hardware available today, you should probably just upgrade yours. Anyway sorry for the late reply.

mikeller, 2018-04-17:
@Daniel-1276: Cleanflight 2.3 and later does have FPort.