cleanflight/cleanflight #1936: Cleanflight should support NRF24L01 based receivers

martinbudden, 2016-03-08:
The NRF24L01 is a low cost low power transceiver in the 2.4GHz band. It is used by many toy quadcopters.

hydra, 2016-03-09:
Agreed, it will for some brushed targets when we get to review the code.  There's a few brushed boards that have the NRF24L01, including the Air Nano.  Often though they use the 64kb flash but we can often compile out unreleated code like we do for the CJMCU.

jarno83, 2016-03-09:
Iforce2d had it working some time ago http://m.youtube.com/watch?v=K10qpQ47Q6Y

martinbudden, 2016-03-11:
@jarno83 , yes I was aware of @iforce2d 's work [here](https://github.com/iforce2d/cleanflight/tree/nrf24). My work started as a simple port of his work to the latest version of Cleanflight, but ended up as a total rewrite (I tend to follow the "plan to write your code at least twice" development method, the first time to learn how it should work and the second (or third time) to do it properly. (And then the fourth time to optimise it, if required.)). I then found out that @goebish had a working NRF24 Cleanflight implementation, so I got in touch with him and we decided to merge our pieces of work together.

There were some aspects of @iforce2d implementation that needed to be changed before inclusion into a Cleanflight release:

1. Primarily was the issue that iforce2d used his own transmitter protocol. This meant you had to build your own transmitter to use it (clearly that meant it was not mainstream).
2. It used a port of tmrh20's excellent [RF24](http://tmrh20.github.io/RF24) library. This library is a c++ library and so had to be shoehorned into C form.
3. The iforce2d protocol used a fixed RF channel and address, so was susceptible to RF interference from others using the same protocol.
4. It implemented it's own failsafe handling, rather than using the Cleanflight failsafe handling.

Note the above are not meant as criticisms of iforce2d's work: I think it is an excellent piece of work; rather they are a statement of changes that needed to be made before it could be included in a Cleanflight release.

iforce2d, 2016-03-11:
Yes, the work I did was very user-specific and it was only worthwhile because I already had a home-made transmitter ready to use. I did not submit any of it as a pull request because of the points that @martinbudden mentions.

My goal was to transmit binary data from other sources (eg. GPS coordinates) as well as the normal control inputs, so it was never going to be a good fit for an off-the-shelf radio.

I think the software SPI driver part of it might be worth including in cleanflight separately though.

martinbudden, 2016-03-12:
Yes, I think the soft SPI driver could be useful. I'll look into this when I get a chance. I quite like the idea of an SP Racing F3 mini with that icstation NRFL01 long range module you reviewed.

lotelx, 2016-11-11:
I'm thinking also about this issue.
I will suggest to be capable of read MSP messages, like we do with the rx/tx serial UART interface.

Then we can use MSP_SET_RAW_RC message for control.
(And all the rest of messages available, like receive data for telemetry.)

closed 2017-02-25 by hydra

2black0, 2017-07-21:
sorry to up this, 
can we make mesh networking using nrf2401l with many drone? maybe for formation flight or swarm?

lotelx, 2017-07-22:
Even if this issue is closed, Cleanflight not support yet the nrf24L01 chip.
You can use transparent TTL to Wifi module and send  MSP messages instead for control multiple drones. (not in mesh communication, just standard wifi)
I'm using this module:
https://www.banggood.com/Geekcreit-DT-06-Wireless-WiFi-Serial-Port-Transparent-Transmission-Module-TTL-To-WiFi-p-1141047.html?p=JB1919987667201412AN

2black0, 2017-07-22:
@lotelx thanks for your suggestion, btw any idea how to identify the number of drone?

hydra, 2017-07-22:
CF now has support for the NRF24L01 receiver on some targets, see src/main/drivers folder for source.

2black0, 2017-07-27:
@hydra how to wiring it? I was googling there is no reference or document describe it,

Marineserver, 2018-12-28:
hi can any one post how to wire nrf24l01 with clean flight assume if i used naze32 with
Receiver Mode = SPI RX Support 
SPI Bus Receiver Provider = NRF24_XXX any one of them

i can't find anywhere about the detailed instruction for connecting..

Thanks in Advanced....

mikeller, 2018-12-29:
@Marineserver: I don't think Naze has an SPI bus exposed, so you'll have to solder directly to the MCU pins.

Marineserver, 2018-12-29:
> @Marineserver: I don't think Naze has an SPI bus exposed, so you'll have to solder directly to the MCU pins.

Hai mikeller thanks for your kind reply, and for your kind reference i have been developing naze32 using stm32 BLUE PILL Mini Development Board... **Reply most welcomed on Wiring NRF24l01 and SPI Flash Support , complete includes(CE, CSN, SCK, MISO, MOSI)**  
Thanks in Advanced.

Cosmin82, 2019-05-05:
@Marineserver there are 2 posible solutions  for that : 1 go with i2force coding and use the softspi or if you are up to the task take the CF 2018 source witch is the last working on naze and recompile folowing i2force guide lines on his github. It will work any way . 
@hydra it would be nice if you can add some "experimental" branch to your CF were crazy PPL do crazy stuff. F1 - F3 are easy to find and lot's of ppl can test. The feedback is what you gain . Think about APM - 8bit chip but is used in 2019. Why? 😗