cleanflight/cleanflight #1944: Advancement for Fixed Wing, Airplane / CF as a Gyro Stabilizer.

badfrogg, 2016-03-11:
I would like to offer some observation and recommend some advancement of features focused on fixed wing applications focused on CL as a gyro stabilizer.

I have successfully (meaning not crashed) used Cleanflight 1.12 in two very different fixed wing airplanes.
1. a 40" Sukhoi SU-31 Acrobatic/3D aircraft with 400w of electric power.
2. a 50" 70's era pylon racer with a .49 glow engine. (Think ultra Stick)

My deployment of the CF has used Naze32 Ver. 5 platforms and "MWREWRITE" controller. The focus has been towards pure flight performance. I only setup Pass-through mode and when using the FC, it is only using Acro mode. I do not even set Level or Horizon as options. My testing has been to develop an understanding of the PID's, Rates, RC Rates, and Mixes as it pertains to servos moving control surfaces on an airplane.

At this point I have found something that i think limits the viability of CF as a fixed wing gyro stabilizer.

Fixed PID Gains are only appropriate for a narrow range of speed. As a plane increases speed the gains need to be decreased. TPA is not really a viable solution since, in an airplane, throttle is not directly linked to airspeed. (full throttle in a climb or hover can equal 0 airspeed) (zero throttle in a dive can be 100mph airspeed).

Possible solutions:
1. PID attenuation on a AUX channel instead of being locked to throttle.
I was going to play around with this by removing the throttle servo / ESC from the flight controller all together and plugging it directly into the RC receiver. Then assigning the throttle channel in the FC to a slider on the transmitter. This is a work in progress to see if i can achieve PID attenuation disconnected from throttle.

Note: I would like to understand what happens to the PID algorithm when the throttle is reduced to zero. I notice that it resets or something. When landing i cut throttle the plane snaps its attitude a little.
 
2. In flight selectable "Profiles" not just "Rate Profile"
Right now you can select rates in-flight on a switch. It would be helpful to be able to select profiles too. For example (Low Rates / High Gain) - for landing or cruising, (High Rates / Med Gains) - Acrobatics, (High rates / High Gains) - for 3D (Low Rates / Low Gain) - Speed runs.

Another issue that i see is that there really isn't a PID "Gain" per say. Im not quite sure how TPA does it but if you reduce the P in PID to zero you will have zero control. If there was a  way to attenuate the effect of the stabilization by means of an overall gain that would be useful. For instance a "stunt mode" setting that attenuates the PID gain depending on the stick position. For example, when doing a flat-spin maneuver the sticks are held at maximum, the gyro is effectively disabled or attenuated to near zero. You let go of the sticks and the gyro stabilization returns.

I hope that all my efforts and testing is useful to the development of CL. I am a good test pilot but not a very good programmer.

mikaeltulldahl, 2016-03-19:
another way to handle the attenuation could be to introduce a speed observer ( observer in the control-engineering sense) which either uses a pressure difference sensor or using a model of the airplane taking pitch angle and throttle into account.

senosahisnu, 2016-03-30:
This is very interesting for me, although until now I used cleanflight exclusively for small quadcopter, since I'm an aerospace engineer. It is true that PID values that works in low speed becomes quite too aggressive in hi-speed flight. One of the solution are using gain scheduling which "schedules" controller feedback by input of flight parameters and environment. Since we don't have airspeed sensor yet in cleanflight (I guess?), maybe accelerometer can be used to "sense" current airspeed by integrating acceleration into velocity. I may adds PID attenuation feature by using accelerometer data. I'll let you know if there's any progress.

ledvinap, 2016-03-30:
@senosahisnu integrating acceleration won't work - drift caused by noise and offset errors will be to high.

badfrogg, 2016-03-31:
Airspeed feedback via an aux channel is totally possible with FrSky. Using a pitot tube sensor to receive airspeed telemetry you can use a lula script to translate that data into aux channel to send back the the flight controller. Easy!

What we still need is a way to attenuate the PID with an aux channel!

closed 2017-02-25 by hydra