cleanflight/cleanflight #1946: TASK_ACCEL is run at 100Hz

ledvinap, 2016-03-12:
Sampling accelerometer at 100Hz causes heavy aliasing (anything over 50Hz is aliased). 
Internal ACC lpf is set to 188Hz (neccesary for good gyro performance), making things even worse.

BARO mode is heavily affected, IMU leveling is probably degraded too (most acc samples get discarded, remaining ones are low quality)

ledvinap, 2016-03-12:
https://github.com/cleanflight/cleanflight/blob/master/src/main/scheduler_tasks.c#L62

digitalentity, 2016-03-12:
@ledvinap you noticed the aliasing as well. With 100Hz accel sampling we don't have a reliable accelerometer readings. IMO acc should be samples along with gyro.

borisbstyle, 2016-03-12:
I experimented with 1khz sampling + filtering but just 100hz processing and
that really helps

digitalentity, 2016-03-12:
@borisbstyle this problem was encountered in iNav right after scheduler was introduced. Solved by reading and processing accel along with gyro. Your approach might be even better - read and accumulate acc at 1khz, downsample and process at arbitrary rate. Great idea.

borisbstyle, 2016-03-14:
I have something like this

# status
System Uptime: 11 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:27%
CPU Clock=72MHz, GYRO=MPU6000
Cycle Time: 130, I2C Errors: 1, config size: 1376

# tasks
Task list:
0 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 156us, avg: 90us, rate: 7633hz, total: 11sec
4 - (SERIAL) max: 762us, avg: 2us, rate: 99hz, total: 11ms
5 - (BEEPER) max: 13us, avg: 0us, rate: 99hz, total: 10ms
6 - (BATTERY) max: 13us, avg: 0us, rate: 49hz, total: 2ms
7 - (RX) max: 102us, avg: 70us, rate: 49hz,

ledvinap, 2016-03-14:
@borisbstyle : So it is safe to increase ACC frequency to 1kHz ? On NAZE?
I can compile the binaries, but can't test it ...

martinbudden, 2016-03-15:
@ledvinap , @borisbstyle , @digitalentity , although the scheduler interface doesn't currently support it, another approach would be to change the `desiredPeriod` of the `TASK_GYROPID` and the `TASK_ACCEL` tasks when the user changes from rate mode to horizon or angle mode (and vice versa).

digitalentity, 2016-03-15:
@martinbudden  
The problem is that when we switch from RATE to ANGLE we already need a correct attitude estimate which is unlikely with ACC running at 100Hz.

borisbstyle, 2016-03-15:
@ledvinap It is save once you dont use softserial, luxfloat and some other fancy features. 

@martinbudden That could be done.....only not sure how the ESC firmwares would be dealing with changing refresh rates. I think that just 1khz sampling + fitlering of acc data and than much slower processing of it would solve it all.

ledvinap, 2016-03-15:
@martinbudden : IMO the consensus is that reasonably accurate position estimate is necessary when switching to horizon/angle mode. Users use this as failsafe, there isn't much time ... 

Filtering  ACC will introduce delay that will affect leveling accuracy (it is possible to rotate gravity vector first and then filter it, so effect of rotation+delay is minimized/avoided. Current IMU implementation roughly does this), but this effect should be quite negligible.
Frame noise suppression is much more important.

Maybe use 1kHz IMU on targets with HW floating point and use lower update rate on Cortex-M3.

ledvinap, 2016-03-15:
It would be great to separate device driver from rest of processing.
Device driver would register into scheduler at desired rate and provide different kinds of data.

So MPU6050 will resister at 1kHz (or INT pin) and provide GYRO, ACC and TEMP. another tasks can trigger when new data are available. This will also avoid double-initialization of multi-function chips ...

This will also simplify using of FIFO on peripheral (8kHz gyro rate may be possible)

martinbudden, 2016-03-15:
Yes, I didn't particularly like the idea of changing scheduler timings, I thought I'd just add it as a suggestion though.

@borisbstyle , your code that samples ACC at 1kHz and processes it at lower rate, is that anywhere near ready for a PR?

@ledvinap , I agree it would be good to separate device driver from rest of processing. I think we'll get to that point eventually.

borisbstyle, 2016-03-15:
@martinbudden 
Its all proof of concept, but @digitalentity is usually the one doing code around level stuff.

FIFO would be nice, which would gain efficiency on SPi targets. Still need to find out what it does on i2c.

ledvinap, 2016-03-15:
@borisbstyle : Main loop can stay at 1kHz, but gyro sampling at 8Khz. 8 samples will be stored in FIFO and read out in single I2C transaction (saving some bandwidth). Also Loop jitter won't affect sample readout (you won't miss sample when called 125us too late).

This effect will be greatly augmented when I2C communication can be done in background ...

borisbstyle, 2016-03-15:
@ledvinap 
That would be the most ideal indeed! I still didnt manage to get to play with FIFO. Many different drivers make it difficult.
Increasing sampling frequency has a surprisingly nice effect on filtering as the filtering delays get lower due to shorter intervals between samples. Besides that it gets as clean as it can get.
Practice showed that up to 500-600hz there is noise visible on gyro. That would mean x10 sampling rate to prevent any possible aliasing artifects, which is essentially 6-8khz sampling.

borisbstyle, 2016-03-15:
@ledvinap 
What kind of scares me on FIFO is overflow handling. I am not experienced with it, but willing to play

borisbstyle, 2016-03-15:
The ideal solution.

Read on 1khz speed. At same time Gyro FIFO buffer (8 samples) + 1 Acc sample and that for each axis ant than process that and do all stuff

ledvinap, 2016-03-15:
Yes, something like that.
FIFO overflow should not be a problem - just flush FIFO if something unexpected happens ...
With MPU6050 there is small problem that partial measurement may be written in FIFO, butreading FIFO length again solved it (it takes long enough for write to finish)

ledvinap, 2016-03-15:
What about this issue? Is it ok to increase ACC task rate?

this issue is show-stopper for some users. We may work on better solution later ...

flounderscore, 2016-03-15:
What about using a ring-buffer to store the samples instead of a FIFO?

borisbstyle, 2016-03-15:
@flounderscore 
FIFO is in the MPU. You need that to prevent multiple reading

@ledvinap 
yeah temporary solution is to increase the task rate. But it is really easy to add code / new task where ATTITUDE estimation is separated from ACC sampling.

ledvinap, 2016-03-15:
It is already separated (possibly by mistake):
```
taskUpdateAccelerometer 
 - imuUpdateAccelerometer
  - updateAccelerationReadings
    - acc.read();
    - convertRawACCADCReadingsToInternalType
    - alignSensors
    - performAcclerationCalibration
    - performInflightAccelerationCalibration
    - applyAccelerationTrims
```
Processing (and RC filtering) is in imuCalculateEstimatedAttitude :
```
taskMainPidLoop
 - imuUpdateGyroAndAttitude
   - imuCalculateEstimatedAttitude
    - accSmooth = filterApplyPt1
```

So increasing update is probably all that is necessary, it just needs some testing  ..

danielgpalmer, 2016-03-17:
Is this issue causing horizon mode to be broken? I upgraded to the current release from an older release recently and it doesn't fly very well at all. If I fly like it's in acro mode and constantly correct the attitude myself it's ok but if I don't it takes it takes a long time to level itself. It seems like the default strength of the leveling was changed recently but I can't get it to fly like it used to even if I ramp it right up.

ledvinap, 2016-03-17:
@danielgpalmer : Accelerometer noise effects are much more noticeable with 1.12. It does affect BARO, HORIZON and ANGLE modes.

It should not change leveling strength, but estimated horizon may be off and it may take longer for it to stabilize.

fifteenhex, 2016-03-17:
@ledvinap What I think I experience is that flying slowly the leveling is ok but if I pitch forward at speed and then release the stick it doesn't manage to level before it smashes into the ground. I think I found if I increased the leveling strength releasing the stick would cause it to level but it was too strong and it would wobble around like jelly before settling. Anyhow based on your comment I'll try an older release this weekend. Thanks :+1:

borisbstyle, 2016-03-17:
@ledvinap
I see in your local branch that you have FIFO going.
I have a few questions.

- Can you make a PR for that as your local branch differs a lot from cf
- Did you measure the transaction times over i2c for reading 8 gyro samples
+ acc
- Your code is applied to MPU6050 driver .....IMO it should be general for
entire MPUxxxx familly.

I actually tried to enable it, but didnt manage to get it going somehow

ledvinap, 2016-03-21:
@borisbstyle : My branch diverged a little too much, I need to incorporate upstream changes / resolve hidden conflicts.
Copying only FIFO interface and modifying driver to do filtering (so one sample per looptime interface is unmodified) is probably easiest now.

The i2c transaction is roughly: 
`<1 addr><1 reg><2 FIFO len> / <1 addr><1 reg><48 gyro> / <1 addr><1 reg><6 ACC>`

~62 bytes * 9bits = 558bits + 3 start + 1 stop
Minimum I2C frequency is ~ 560kHz (assuming full I2C utilization).
At 900kHz it will take ~630uS (63% of available time). It is quite possible that remaining 37% will be enough to do computations on F3 targets, but it would be much better to implement I2C in background ...

kgustine, 2016-03-24:
Will this be added to the 1.13 milestone?

I haven't been experiencing any issues with normal flight, but the more flips I do, the more the level orientation gets off.  I'm hoping this will help it...

digitalentity, 2016-03-24:
We won't be able to solve level drift without sampling accelerometer faster.
I think background I2C is key to solving the issue now.

ledvinap, 2016-03-24:
@digitalentity : 1kHz ACC readout shall be possible with current driver.
Testing it is all that is needed (I don't have working NAZE right now to do it)

calvarez, 2016-03-24:
I have a 450 with a Naze32 Full on it and wouldn't be averse to testing, however I haven't worked with testing CF before.  I'm a Linux admin and can find my way around things ok, but have no idea how to put this together from source.

martinbudden, 2016-03-24:
@calvarez , have a look at [Building in Ubuntu](https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Ubuntu.md) for instructions on how to build Cleanflight. The [development](https://github.com/cleanflight/cleanflight/tree/master/docs/development) directory has instructions on how to build for other platforms as well.

martinbudden, 2016-03-24:
@ledvinap , I've done a bit of testing running the ACC at 1kHz on a Naze derivative (Micro SciSky brushed quad) and haven't had any problems yet. More testing required though...

calvarez, 2016-03-24:
Ok, so all the current sources here have the 1k refresh?  I may not be able to get it tested today, but certainly by the weekend.

martinbudden, 2016-03-25:
@calvarez  the refresh rate is defined in the `scheduler_tasks.c` file, see:
https://github.com/cleanflight/cleanflight/blob/master/src/main/scheduler_tasks.c#L59-L64

You'll need to set `.desiredPeriod` to 1000 (i.e. 1000 microseconds, so 1kHz) before rebuilding.

calvarez, 2016-03-26:
Got everything working to build it, but ran into a question.  That setting has two numbers.

.desiredPeriod = 1000000 / 100,     // every 10ms

So I'm not sure which one should be changed to 1000.  I believe the last one, but since this could have serious consequences for my quad, I'd like to be sure.

calvarez, 2016-03-26:
I did more reading and figured out that it is indeed the second number.  Changed it to 1000, and success!  The problem I had where the quad would change altitude randomly and harshly is gone.  I did not tune any other parameters like the filters.  I did a full erase and flash, then only set up the parameters necessary for my quad such as SBUS, cycle time, and the like.

martinbudden, 2016-03-26:
It's good that it seems to solve the problem. Let us know if there are any problems that might be attributed to the higher ACC refresh rate.

I've been testing at 1000 as well, and haven't yet encountered any problems.

calvarez, 2016-03-26:
I just did a 15 minute flight outside with some wind gusts (5-10 MPH), and things are vastly improved.  My PID is still all wrong, but now I can start tuning it.  Before this, I was getting crazy altitude changes that made it impossible to tune anything.

I didn't notice any downside to this.  All other behaviors seem the same.  Just no more jumping.

Also, I am NOT using baro mode, this is just normal flight in both acro and horizon modes.  I will try baro later when the wind dies down.

martinbudden, 2016-03-27:
@borisbstyle , @ledvinap , @digitalentity , it seems like changing the ACC refresh rate to 1kHz is unproblematic - at least good enough to go into betaflight, if perhaps not yet Cleanflight.

By the way, I've added a page to the DISPLAY code to display task statistics, see https://github.com/martinbudden/cleanflight/commit/f1928b3a202678f210e65859bb3214bb401ce165
feel free to use it if you think it is useful. Unfortunately my OLED display is partially broken (it displays only ever other line), so I'm not 100% sure that my display code if correct, but from what I can see on my broken display it looks OK.

sethyx, 2016-03-30:
I just did a quick test on a Naze rev6 (I also had the drifting issue over time) - the difference is visible from blackbox logs even on a table without props.
The second log was made with cleanflight v1.12.0 + the aforementioned one-liner change.

[ACC_100Hz.png](https://cloud.githubusercontent.com/assets/1311824/14156346/3c9bec46-f6c6-11e5-8a87-83b2820e108e.png) - highest peak-to-peak amplitude = 1g
[ACC_1KHz.png](https://cloud.githubusercontent.com/assets/1311824/14156347/3ca13e30-f6c6-11e5-8292-76fd70eb76d6.png) - highest peak-to-peak amplitude = 0.2g (and the pitch/roll drift is also gone)

Blackbox logs, if anyone is interested: [bbox_acc.zip](https://github.com/cleanflight/cleanflight/files/196392/bbox_acc.zip)
I'll try to check if there are any downsides in the air over the weekend.

hydra, 2016-05-07:
should be fixed by the merge of #2076

closed 2016-05-07 by hydra