cleanflight/cleanflight #1947: gyro calibration in cleanflight 1.12

wildflyer, 2016-03-12:
stick command for gyro calibration does not work anymore in 1.12 on my naze32 rev5.how can I manually calib gyro before start  ?

borisbstyle, 2016-03-14:
It does work with stick command

wildflyer, 2016-03-14:
hi,

stick commands seems not to work on my quad anymore, it flies well, but stick commands does not work, what am I doing wrong ?
I have 2khz, acc disabled

greetings

joachim

borisbstyle, 2016-03-14:
does your rx input reac min_check and max_check?
That is the only way that it would not work

wildflyer, 2016-03-14:
I just tried that, values are 1100 and 1900 min max. tx values sent are 1000 and 2000 min/max.

I also noticed since I flashed 1.12, that on startup there ist no gyro calib there anymore, too, in v1.9 it used to "wait" until the quad is stable on the ground, and then performs gyro calib. this is no more in v1.12 ?

wildflyer, 2016-03-16:
I found out,
stick command works, but the escs are not beepeng anymore after calibration, so I thought it would not work. the same on startup. how can I activate to "wait until no movement" before gyro calibration on startup, as it was in cf 1.9 ? why did this change ?
greetings
joachim

closed 2017-02-25 by hydra