cleanflight/cleanflight #1949: The setting for acc_lpf_factor is gone?

calvarez, 2016-03-13:
I've been trying to set this in 1.12, and it tells me that's not a valid name.  I need LPF to be more than 4 for my build.  Yes, props are balanced and the Naze32 is in an isolation mount, but it still does a weird vertical drop/climb randomly.  Setting this to around 20 fixes it.

calvarez, 2016-03-14:
Can anyone tell me if they are still able to get/set this parameter on a Naze32?

foot596, 2016-03-15:
Looks like it was replaced with **acc_cut_hz** here 1e6dfc925f7cc663b89f790bab97c9c239f699fd
```
+    uint8_t acc_cut_hz;                     // Set the Low Pass Filter factor for ACC. Reducing this value would reduce ACC noise (visible in GUI), but would increase ACC lag time. Zero = no filter
```

calvarez, 2016-03-16:
I see old forum discussions about this setting, so it seems like it's not new?  It was defaulted to 15 (old LPF default was 4).  I tried 5, 40, and 100 without any improvement.  I get vertical instability, and a little bit of horizontal.  The FC is on a silicon/neoprene vibration isolation mount.

lainchir, 2016-03-16:
What size is your quad Calvarez? The reason I ask, is because my 450 had a vibration isolation mount, and it flew awful on CF 1.11. All motors were well balanced and props/hubs balanced too. While I was still 1.11, I removed the mount, and stuck the Naze32 down with ordinary gyro foam tape and it totally transformed it. Sadly it's not good again under 1.12, but I'm hoping stability will return when the acc/gyro issues are fixed.

calvarez, 2016-03-16:
It is a 450 also.  I first had issues with vibration and the Naze, then used a few layers of foam tape and it was good.  It still needed the acc_lpf_factor set a little higher than default.  Then I got the vibe mount and it was MUCH better, though it still needed acc_lpf_factor set.  Then I put in 1.12 and it was all over the place.  Did a lot of screwing around, and couldn't figure out why.  So I defaulted the Naze, went to do it all again manually, and found this setting was missing.

lainchir, 2016-03-16:
I currently have mine rolled back to 1.11 due to instability. Waiting, fingers crossed for the guys to do their magic. Hopefully, this won't become a "race only" firmware

calvarez, 2016-03-16:
I had other issues that 1.12 was supposed to fix.  Maybe I just need to go back to Baseflight?  Dunno.  I've got a dozen hours into trying to make this quad fly straight.  About ready to just order an APM 2.8 and do something totally different, just not sure.

calvarez, 2016-03-17:
Yesterday we figured out that gyro_sync now defaults to on, and locks the looptime to the gyro's speed.  This was causing the ESCs to flip out.  Turning that off puts the loop time back to a proper level, and the majority of the craziness is gone.  Not all, but 90% improved.  Nothing in the UI gives an indication of this problem other than looking at the actual reported cycle time.  And since the UI uses "cycle time" for actual and "loop time" for settings, I wasn't sure if they should match or not.

lainchir, 2016-03-17:
I switched my gyro_sync to off, and set my sync_denom to 2 and it still flies better with 1.11. I notice that there's also an ALT PID doc thats been uploaded. I'm happy to go to 1.12+ and do tests/blackbox logging, but on my 450 running 1.12, flight at default isn't good. Hoping a new release will fix it tbh.

Cycle time and looptime... in 1.11 I'm flying with 2500 looptime. What does this mean when the UI uses cycle time? Am I not adjusting looptime anymore? It seems that there's been so many changes to the way this is configured in one revision (I didn't use the RC releases)

commit e133331c (2016-04-07): Update CLI documentation for acc_cut_hz.  Closes #1949
commit a037ad6b (2016-04-10): Update CLI documentation for acc_cut_hz.  Closes #1949
closed 2016-04-07 by hydra