cleanflight/cleanflight #1961: SP Racing f3 no response from bootloader

ATXorion, 2016-03-17:
Trying to calibrate a new ESC and i cannot connect to cleanflight on my Mac (i installed the drivers).

This is a new Storm V3 drone  with the SP F3 and i have not connected it to cleanflight before.

Upon connected via USB the FC LEDs flash and beeps and says "No configuration received within 10 seconds, communication failed"

I have not tried to jump the bootloaders; i really do not want to flash the preset tuning on the quad.

(note- i did try to flash and bootloader did not respond).

cruisetheblues, 2016-03-18:
I had a similar problem. There are a few workarounds that you can try. First try using a manual baud rate of 256000 or 115200. If that does not work you will need to jumper the boot pins.

ATXorion, 2016-03-18:
Thanks. I tried all remedies and am sending back the drone to be replaced.  Helipal agreed to exchange it as the FC has issues and is defective.  

I tried all baud rates- 5 different USB cords- 3 different computers- jumping the bootloaders and other stuff.  No luck connecting to cleanflight. 

The drone did fly great until an ESC went out.  I replaced the ESC and couldn't calibrate it in cleanflight.  

Thanks for the response.

SteveAmor, 2016-03-18:
@ATXorion - if this issue is resolved can you please close.

closed 2016-03-19 by ATXorion