cleanflight/cleanflight #1962: Allow configuration of in-flight adjustment function application rate

timdonovanuk, 2016-03-17:
Currently "while the adjustment switch held is high/low, the adjustment function applies and increases/decreases the value being adjusted twice a second" - this results in a bit of a crap shoot as to exactly how much you increased/decreased a value. 

It would be great to have an option to:

a) completely disable the repeat application of the adjustment function when an aux channel is high or low. Simply put, even if a switch was held high or low, it would only adjust the value ONCE, until the switch was placed back to middle. 

b) Introduce something on the adjustment screen that says 'if enabled'..'when channel is in rage'...'then apply'...'BIG' or 'SMALL' changes (checkbox). There is better terminology but I cannot think of it. Aagain, simply put, this would allow you to then decide if you want the increase/decrease functionality to be as it is currently, or as I am proposing.

Why am I proposing this? Well it occurred to me, using a LUA script on a Taranis, and seeding the script with your *CURRENT KNOWN PID VALUES* it would not be too difficult to provide an on-screen display of PID adjustments using the current adjustment mechanism. However, its currently not feasible, because if a user holds a switch in the high position for 4.78 seconds, it's not easy to calculate how much the value may have been increased by.

Thanks!

joeynuggetz, 2016-03-29:
This sounds interesting. I'd like to see this implemented too.

fftunes, 2016-04-01:
for b) 
why not let us directly choose the value for each step. Instead of vague description we could simply put a number. Or maybe a selection table like 1 / 10 / 100 units?

closed 2017-02-25 by hydra