cleanflight/cleanflight #1964: Channel Mapping Incorrect

AndrewV1, 2016-03-18:
Hi Chaps
SP Racing F3 Flight Controller board not mapping the correct channels when using the XBus port set up on a JR Prop RG712BX  XBus Receiver. Running in Xbus mode B.in serial mode. 
Controller board                                     Receiver
Roll                       is swapped with          Pitch (Elevator) 
Throttle                 is swapped with          Roll (Aileron)
Pitch                      is swapped with         Yaw (Rudder)
Aux1                      is swapped with         Throttle
Yaw                       is swapped with          Aux1

All settings and videos on this controller were followed carefully. I have flashed it Three times to make sure it is correct. Channel Mapping is set to TAER1234. UART3 set to Serial RX.
Is there any other settings I am not aware of ??? I am out of clues as what to do.......
Thank You for the help in advance.
Cheers
AndrewV1

StraTech, 2016-03-27:
Hi There,
SP Racing F3 Flight Controller board not mapping the correct channels  in PWM mode either, I'm unable to get any RX input to IO-1 (CH1,CH2) & IO-2 (CH3,CH4). But I can get all AUX CHannels!
The ports dont seem to be turned on or mapped properly! Help!! Running 1.11 Firmware.

AndrewV1, 2016-03-29:
StaTech. On the Clone PCB's the colors of the wire looms differ in color as to the Original Clean Flight PCB's
Mapping in PWM Mode:- 
IO1:-                                                     IO2:-
1= Black - 0v                                        1= Black - 0v
2= Red - +5v                                        2= Red - +5v
3= White - Throttle                               3= White - Elevator (Pitch)
4= Blue - Ailerons (Roll)                       4= Blue - Rudder (Yaw)
5= Yellow - AUX1                                  5= Yellow - AUX3
6= Green - AUX2                                  6= Green
7= Red 2                                              7= Red 2
8= Purple                                              8= Purple
I Flashed the Board to Version 1.12.0 Firmware. I hope this helps you out. I am still waiting for some sort of Answer from the Team. They want our support but they do not let one know as what is wrong with the Mapping in Xbus mode..... I was even hoping for someone to tell me to get lost ..... then at least I know were I stand on the matter....... I would have liked them to make a manual setting up of the Channel Mapping in ALL Modes. That way you can set it up no matter what radio you are using. Maybe I will be lucky for a response in a few weeks time ......................!!!!!!!

hydra, 2016-04-07:
This is not the place for user support.  Try rcgroups.com threads.

closed 2016-04-07 by hydra

StraTech, 2016-04-08:
I thought there was genuine bug with the code, an incorrect picture was the cause!
There isn't many pinout pics and I didn't realize on SPF3 pins 1-8 on IO-1 were reversed on IO-2 side (8-1), but it was strange that the configurator showed no input on CH1 & CH2 JUST because CH3 & CH4 weren't wired up. P.S. I wasn't looking for user support here.......but hey! thanks anyway & I'm glad to see a response!

hydra, 2016-04-16:
yes, it'd be good to show the user the raw rc inputs as well as the FC internal state on the receiver tab to better illustrate the current configuration settings/wiring/mapping, etc.