cleanflight/cleanflight #1967: Header files should include their own dependencies

Windfisch, 2016-03-20:
Hi,

while trying to use portions of cleanflight-code in another project, I noticed that cleanflight's `.h`-files require you to include their dependencies *manually* before including them.

As an example, the `src/main/sensors/acceleration.h` file uses typedefs like `uint16_t` and `acc_t`, but without ever including `stdint.h` or `drivers/accgyro.h`. Instead, it relies on the unitfile to include the relevant files *before* including the `sensors/acceleration.h`, requiring any developer to get the ordering right.

Instead, I'd like to see every header file to `#include` their dependencies on their own, for the following reasons:

1. It improves the reusability of code in other projects
2. It explicitly states the dependencies
3. It's easier for new developers (who tend to look only at the `.h`-files/declarations to get "the big picture" first, before they start digging in the implementation files) to figure out where what comes from

For the sake of reproducability:
Expected behaviour: in `src/main/`, running `gcc sensors/acceleration.h` should run without errors and generate a precompiled header file
Actual behaviour: I get a lot of `error: unknown type name` messages

In this particular case, the solution would be the following diff:
```diff
diff sensors/acceleration.h sensors/acceleration.h
--- sensors/acceleration.h
+++ sensors/acceleration.h
@@ -17,6 +17,11 @@
 
 #pragma once
 
+#include <stdbool.h>
+#include <stdint.h>
+#include "sensors/sensors.h"
+#include "drivers/accgyro.h"
+
 // Type of accelerometer used/detected
 typedef enum {
     ACC_DEFAULT = 0,
```

If you agree with me, I'd offer to try to add the necessary includes semi-automatedly.

jflyper, 2016-03-21:
Once upon a time, I thought of the same thing about config_master.h; see #1890 for more info.

Windfisch, 2016-03-21:
Referring to hydra's explanation in #1890: While it is true that you see what the dependencies are in the `.c`-files, it makes it completely unclear what the dependencies are when looking at the header files. (And the headers is usually the first place for a new developer starts to search.)

By the [https://en.wikipedia.org/wiki/Design_by_contract](Design by contract) principle, a mere *user* of the functions in a unitfile should not be forced to actually read and understand their implementation; instead, they can get all information they need from the `.h`-files. In its current form, this is not true, because any user still needs to skim through the implementation files to sort out the dependencies manually.

In the particular case of `config_master.h`, I was thinking about not editing this file out of similar reasoning; however, all the other header files definitely should carry their own dependencies IMO.

bdoiron74, 2016-03-21:
I agree. If a c file only wants to use a particular module, requiring all
the dependencies of its header file to be placed in the c file makes no
sense. When something gets added to a module that creates a new dependency,
you have to visit every c file that includes it to add that dependency -
even if they aren't using the new feature. Gross. Having to manually order
your inclusions just right is pretty atrocious too.

--Bob

hydra, 2016-03-24:
not yet. though we may revisit this later.

there any also many outstanding pull requests that would break if we changed this now.

closed 2016-03-24 by hydra