cleanflight/cleanflight #1968: Documentation for current_meter_type value seems to be wrong

XDA-Bam, 2016-03-20:
The documentation lists the values [0, 1, 2] for current_meter_type, corresponding to no sensor/hardware sensor/virtual sensor. Trying to set current_meter_type to one of those values in the CLI results in "invalid value" error with Cleanflight 1.12. The correct values are [ADC] and [VIRTUAL] instead of [1] and [2] (tested). [0] most likely will be [NONE], but I didn't test this one, yet.

Tested on SP Racing F3 clone with Cleanflight 1.12. If necessary, [dump] can be provided.

Pierre-A, 2016-03-20:
**@XDA-BAM**, good catch , thanks. Values are "NONE", "ADC", "VIRTUAL". Can you open a PR for updating the _battery.md_ file ?

XDA-Bam, 2016-03-20:
No, sorry. I don't have GIT set up right now :|

Pierre-A, 2016-03-20:
OK done, see PR #1969 so you can close this issue.
Thanks.

XDA-Bam, 2016-03-20:
That was FAST. Thanks! :)

closed 2016-03-20 by XDA-Bam