cleanflight/cleanflight #1975: Flying Wing servo moves while throttleling

Zomhad, 2016-03-22:
Hi you guys out there. I just finished building a new Mini Racing Wing which I wanted to stabilize with a CC3D flashed with BetaFlight I had laying around. So I put it in and everything looked fine (servos moved in the right direction, stabilization looked good too). But as I was at the field and gave it a little test thrust, I noticed that as I moved the throttle stick, the left servo would start moving upwards with a constant speed till it reached it maximum angle it then stayed there till I zeroed the throttle again. Doesnt matter how much throttle I give, till there is a tiny little bit the servo starts moving.
At home I checked wether I did anything wrong in the configuration tab at the mixers, but everything looked fine to me since I had flying_wing selected. So I moved on to the motors tab to see what's going on with that servo. Now the weird things begin: As I was in the motors tab in Cleanflight with the craft connected the servo didn't move anymore while I was rising the throttle stick. But as I moved out of the motors tab and tried the same thing again, the same problematic happend.
That indicated to me, that the reason for the problem has to be anything with the software and not with the hardware. Because of that I started thinking about what could cause that kind of a problem and first thing that came in my mind was that there is anything wrong with the mixers, so I read this page (https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md) and tried to write my own mixer in the CLI. I'm quiet unsure if mine is working, but that doesn't matter anyway because the same problem happened again, but not only with one servo, now both servos are moving as I'm rising the throttle. 

I'm really done, please help me!
Best Regards,

Zomhad

Zomhad, 2016-03-22:
UPDATE! 
After a bit of thinking, I just decided to don't controll the throttle via the board and let the board only control roll and pitch and don't let it know when I move the throttle stick. Problem was, that I was using PPM so the board still got the information when I rised my throttle, so I decided to go with PWM and wired everything up. I was very happy, because I thought this plan will defenetly work, but I was wrong. The problem I have now, is that the board doesn'st reconise anything I do on the Radio (no RC input is working) BESIDES the AUX 1. But only the AUX 1. Rest is just not working -.-
I thought that my connections might be broken or anything so I tried them out on another board, but they're working great. So it has to be anything with the software. 

I'm stumped
Regards

Zomhad

sppnk, 2016-03-23:
Hi, I think is just an acc calibration issue. I had the same problem with iNav branch and I am now flying well. Good luck!

SteveAmor, 2016-03-23:
You will have more luck asking for support in a forum like RCGroups.  Issues should be raised for bugs in Cleanflight. If you raise an issue for a bug, information like version of Cleanflight code used would help developers find the bug.

closed 2016-03-24 by hydra