cleanflight/cleanflight #1977: Improve blackbox data flash performance

krzysztofmatula, 2016-03-24:
Downloading blackbox log from SPRacingF3 takes looong time. During  that operation the cleanflight configurator shows serial link utilization at ~50%. I assume flash chip is not a bottleneck here and it should be possible to get close to 100% effectively reducing log download time by half.

Another annoying thing is data flash erase. Would it be possible to speed up this operation also?

borisbstyle, 2016-03-25:
Thats not possible I believe. Maybe @thenickdude can explain that better

joshuabardwell, 2016-03-25:
Boris means that the dataflash erase is impossible. I believe there have been several approaches to improving USB serial utilization over VCP, although I don't know what the current status is.

The flash erase is a function of the chip. It's just really slow to erase flash memory, or something.

thenickdude, 2016-03-25:
Yeah the chip performs the erase itself and it always takes 20-30s. The only way to improve it would be if very little data was written to the chip (then only those sectors could be erased instead of the whole chip).

krzysztofmatula, 2016-03-25:
Well, correct... I've just consulted the flash chip datasheet and it says the "bulk erase" time is ~60s...
Implementing sector erase of used parts of the flash has not much sense as the flash tends to fill quickly.

closed 2016-04-06 by krzysztofmatula