cleanflight/cleanflight #1979: Cleanup proposal - rearrange PID code to make it more testable

martinbudden, 2016-03-24:
Currently it is difficult to write test code for the PIDs. This is because the core PID code is not directly accessible. The PID core bases its calculations on the `AngleRate` and the `gyroRate`, but since these are not directly accessible the test code has to spoof these values by setting `rcCommand` values. This is not very satisfactory since, this causes rounding errors and does not really allow precise testing. It also means other aspects of the PIDs are untestable (or very difficult to test).

I propose to rectify this (so that I can write some more test code) by the following:

1) Split the PID low level code into a separate function. The low level code is the code after the line
```        // --------low-level gyro-based PID. ---------- ```
in the PID function. The cost of this is an extra function call, but it makes the PIDs both more readable and more testable.

2) Split each PID into a separate `.c` file, so we would have `pid_luxfloat.c`, `pid_mw23.c` and `pid_mwrewrite.c`, as well as the controlling `pid.c`.

@hydra , if you like this proposal then reply to that effect and I will submit a PR in due course.

digitalentity, 2016-03-25:
@martinbudden, @hydra 
You may want to have a loop at a new PID controller in INAV (see https://github.com/iNavFlight/inav/pull/96). It's currently float-point only, but when the code mature I might convert it to fixed point to improve performance.
It was derived from LuxFloat but now it's completely different. I also did some refactoring that will make unit-testing easier (no tests implemented though).

martinbudden, 2016-03-25:
@digitalentity , yes I'm always interested at looking at different approaches. And the new `HEADING_LOCK` flight mode looks interesting.

I'm flying quite a bit indoors at the moment, brushed quad with a prop guard (actually it's a Hubsan with the flight controller replaced with a Micro SciSky - fits in quite nicely). I'm finding the quad is quite resilient to crashing into walls and the furniture - if you are in angle or horizon mode the quad recovers hitting something and can continue flying. If you are in rate mode, it's more difficult, but you can recover if you flip into angle mode quickly enough. I imagine adding `HEADING_LOCK` would improve crash resilience - so that will be fun for me. I realise that's not what it was designed for though!

I think that adding `HEADING_LOCK` to the existing PIDs, rather than adding a new PID might be the preferred approach though.

digitalentity, 2016-03-25:
I am actually thinking about **leaving** only one PID controller, not **adding** another one.

martinbudden, 2016-04-16:
Addressed by PR https://github.com/cleanflight/cleanflight/pull/2016

closed 2016-04-16 by martinbudden

martinbudden, 2016-04-25:
@digitalentity I've now had time to look at your PID controller and think it addresses some deficiencies in the current PID controllers, in particular

1. The current PID controllers do not properly address ITerm windup
2. The current PID controllers have a use a rather kludgy approach to angle and horizon mode

In addition the iNav PID controller uses a noise-robust differentiator to calculate the DTerm.