cleanflight/cleanflight #1980: Need more MSP support

AlexStudio, 2016-03-25:
I recently built a quad with SPRF3mini and found that 2 MSP is not enough. I'd like to have:

USB VCP => MSP
UART1 => Serial RX for XBus
UART2 => MSP for Bluetooth
UART3 => MSP for OSD

I don't think I'm the only one needs a setup like this. A 3rd MSP is very much required.

martinbudden, 2016-03-25:
There is some precedent, see [the code](https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.h#L258-L259):
```
// Each MSP port requires state and a receive buffer, revisit this default if someone needs more than 2 MSP ports.
#define MAX_MSP_PORT_COUNT 2
```

However, presumably you don't use the USB VCP MSP and the Bluetooth MSP at the same time, so rather than increasing the number of MSP ports, it might be possible to share them (I don't know this area of the code though, so I am speculating).

@hydra, looking further into the future, it would be nice to see the OSD and Bluetooth connected over I2C. This would of course require someone to produce the hardware -  but "build it and they will come". Of course this is a very low priority idea at the moment, but something to think about for in the the future.

Arise, 2016-03-30:
I agree, I would like to see this implemented. I've checked last night for something similar, and I can't get more MSP on more than 2 serials. 

It helps me because I would like to build an app that could send MSP messages directly to the FC, and UART1 is linked with MSP to use as a CleanFlight Configurator and check things live on the screen, but I would need at least one more or preferably two more UARTs so I can connect the apps that sends the MSP messages. Right now I can connect a PC computer with a serial COM port or a Rasbperry PI or an Android phone running EZ-GUI, but I would like to do at least two in the same time.

Also, if I use the UART2 for GPS, that will force me to go using softserial.

Make it possible to enable UART1, UART2 and at least one SOFTSERIAL for MSP.at the same time.

hydra, 2016-03-31:
@martinbudden msp via I2C is pretty much what BST is on the COLIBRI RACE IIRC.

cephyr, 2016-12-22:
Also looking for more than one MSP port on VCP targets, would be a nice enhancement

closed 2017-02-25 by hydra