cleanflight/cleanflight #1988: Telemetry Docs mention ability to run SmartPort on softserial, but seems like baud is too fast.

CptanPanic, 2016-03-28:
According to last paragraph in: https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md you can run SmartPort over softserial, but looking at header for SmartPort.h the baud is 57600, so this shouldn't work over softserial correct?  Are the docs correct?

DzikuVx, 2016-03-30:
Running S.Port at 9600 or 19200 is more less like a hack. I have it working currently with X8R and SPracingF3 and looptime 2000

borisbstyle, 2016-03-30:
That works even with looptimes of 250.

DzikuVx, 2016-03-30:
@borisbstyle not so sure about that. I was unable to setup this with betaflight 2.1.6 and kind of gave up since I did not needed that on that quad

CptanPanic, 2016-03-30:
Looking at the softserial code, it is not even clear if the 19200 baud limitation even still exists.

Ede2016, 2016-04-15:
I had telemetry working on SoftSerial with the AfroMini together with the FrSky XSR or with the X4R-SB. I used 10k resistors from the TX and the RX and then connected the resistors to the SmartPort. Here is what I did: http://fpvlab.com/forums/showthread.php?46100-New-FrSky-XSR-–-3-8g-–-buy-it-now&p=781540&viewfull=1#post781540
This worked with CleanFlight 1.11.0. I could set the baud rate to Auto or anything. It always worked.
Now I upgraded to CleanFlight 1.12.1 and telemetry on my Taranis only discovers the receiver telemetry data like RSSI but not anymore the FC telemetry data.
I never used CleanFlight 1.12.0 so I don't know if that works.

closed 2017-02-25 by hydra