cleanflight/cleanflight #1993: CC3D problem with the gyro on firmware 1.12.0

a6j, 2016-03-31:
Controller CC3D
Firmware 1.12.0
Frame tricopter 250
After updating the firmware to 1.12.0 I have detected a problem. After 3-5 seconds of flight copter starts to pull to the side.
Reset does nothing settings. Turning the controller also does not solve the problem. This issue was on firmware OP. With the transition to the CF this is not a problem only manifested itself on firmware 1.12.0. I downgrade 1.11.0 problem with deviation toward disappeared.
[image]

fftunes, 2016-04-01:
Sounds like your cc3d suffers from vibration, which might become noticeable because of different default filter settings in the firmware?

a6j, 2016-04-03:
Yes vibration little is there, but not critical. How to solve this problem in the firmware 1.12.0?

MrJCub, 2016-04-18:
Are you flying in Auto Level mode?

martinbudden, 2016-04-18:
This looks like a duplicate of issue https://github.com/cleanflight/cleanflight/issues/1946

This will be fixed in 1.13.0. There might be an interim 1.12.2 release to fix it.

a6j, 2016-04-25:
OK. Thank you.

a6j, 2016-04-25:
> Of course in the mode of stabilization. The acro mode, the controller does not control the balance.

closed 2016-04-25 by a6j