cleanflight/cleanflight #2000: smixes don't map to servo ranges

Elmeerkat, 2016-04-01:
When using an smix with a servo that has a range of anything more than 1000-2000 the mix takes the input which can only have a range of 1000, making the servo endpoint commands only useful for limiting the range of the servo, but not extending it.

An example:
servo ranges from 800-2200
input is stabilized roll ranges from -500 to 500 and is currently at 500
servoMixer function calculates servo_width of 1400
calculates min as -700, max as 700
sets the currentOutput to input[INPUT_STABILIZED_ROLL] so currentOutput = 500
currentOutput is multiplied by rate of 125 so currentOutput = 625
currentOutput is constrained between -700 and 700

so the edges can never be reached
fix could be mapping currentOutput to the min and max of the servo which are calculated.

I tried this and it appears to work, though I'm not too proficient with coding, so I'm sure it could be done more efficiently.

hydra, 2016-04-08:
can you show us the code changes you made so we have something to reference please.

Elmeerkat, 2016-04-08:
Starting at line 777:
```
            if (currentServoMixer[i].speed == 0)
                currentOutput[i] = input[from];
            else {
                if (currentOutput[i] < input[from])
                    currentOutput[i] = constrain(currentOutput[i] + currentServoMixer[i].speed, currentOutput[i], input[from]);
                else if (currentOutput[i] > input[from])
                    currentOutput[i] = constrain(currentOutput[i] - currentServoMixer[i].speed, input[from], currentOutput[i]);
            }

            servo[target] += servoDirection(target, from) * constrain(((int32_t)currentOutput[i] * currentServoMixer[i].rate) / 100, min, max);
```

I did this:
```
            int16_t scaledInput = scaleRange(input[from], -500, 500, min, max);
            if (currentServoMixer[i].speed == 0)
                currentOutput[i] = scaledInput;
            else {
                if (currentOutput[i] < scaledInput)
                    currentOutput[i] = constrain(currentOutput[i] + currentServoMixer[i].speed, currentOutput[i], scaledInput);
                else if (currentOutput[i] > scaledInput)
                    currentOutput[i] = constrain(currentOutput[i] - currentServoMixer[i].speed, scaledInput, currentOutput[i]);
            }

            servo[target] += servoDirection(target, from) * ((int32_t)currentOutput[i] * currentServoMixer[i].rate) / 100;
```
adding that variable allows you to remove the constrain on the final addition to servo's output
It appears to work in my configurator, on the Naze mini, but I'm not the greatest coder so you may want to check it.

closed 2017-02-25 by hydra