cleanflight/cleanflight #2006: PID DTerm should always be calculated from measurement, not error

martinbudden, 2016-04-03:
Currently there is a setting `pidProfile->deltaMethod` that allows the PID DTerm to be calculated from measurement (`DELTA_FROM_MEASUREMENT`) or from error (`DELTA_FROM_ERROR`). The current default setting is `DELTA_FROM_MEASUREMENT`.

According to [PID Control and Derivative on Measurement](http://controlguru.com/pid-control-and-derivative-on-measurement/):

> The derivative of e(t) [error] is mathematically identical to the negative of the derivative of PV [value] everywhere except when set point changes. And when set point changes, derivative on error results in an undesirable control action called derivative kick.

There is no advantage to ever using `DELTA_FROM_ERROR`, and there is a disadvantage.

So the `pidProfile->deltaMethod` configuration should be removed and DTerms should just be calculated using the current `DELTA_FROM_MEASUREMENT` code.

ledvinap, 2016-04-03:
IMO D-term from error will boost response to setpoint change. Using output will dump response.
At least with RC filtering setpoint change is smooth, so D-term may improve performance.

Can someone compare flight experience?

martinbudden, 2016-04-03:
@ledvinap , I very much doubt anyone uses this. It's only settable from the cli, it's not described in the documentation, so you'd have actually had to have read the PID code to understand what it does.

ledvinap, 2016-04-03:
If I remember correctly, this code was designed to help with RC steps - as an alternative to RC filter. It help a lot when there are steps in setpoint value. But it will degrade performance to some extent with ramped RC signal (rcSmoothing enabled).
I'll try some simple simulation tomorrow ...

borisbstyle, 2016-04-03:
@martinbudden 
You are probably right, but as the earlier method was from error I decided to keep it configurable in case someone would miss it.
And @ledvinap is right as the from error method can get a bit sharper stick response, but there was just too much discussion about dkick causing confusions that we decided to move it to from measurement method. Till now no complaints :D

martinbudden, 2016-04-03:
@borisbstyle , well if nobody is complaining, then I'd just get rid of it! It makes the code just that little bit easier to read, and it saves a bit of ROM as well (mainly in the cli).

borisbstyle, 2016-04-03:
Well that was the initial thought.....keep in there see if someone complains.....if not than remove several versions later

martinbudden, 2016-04-16:
Addressed by PR https://github.com/cleanflight/cleanflight/pull/2016

closed 2016-04-16 by martinbudden