cleanflight/cleanflight #2020: G-Tune doen't work on 3D Option

Pomaroli, 2016-04-07:
Hi,
i habe a F450 3D Copter moutning a Flip32 FC with Firmware Cleanflight 1.12 and 3D option enabled.

Here is the procedure i used, but i'm not getting any change of PID's.

1) Plug in flight battery [DISARMED, GTUNE off]
2) Switch on G-Tune mode [DISARMED, GTUNE on]
3) Arm the copter [ARMED, GTUNE on]
4) Lift off slowly, avoid stick inputs (Roll and Pitch)
5) Eventually the copter should fly well... when done:
6) Turn off G-Tune while flying [ARMED, GTUNE off]
7) Land and disarm copter [DISARMED, GTUNE off]
8) Store G-Tune P values [DISARMED, GTUNE on]

hydra, 2016-04-08:
thanks for the report.

@MJ666 do you think it should work in 3D mode?

MJ666, 2016-04-08:
One common issue is if the RC input is to noisy the algorithm will not start to tune at all. Roll and Pitch need to be stable at mid-stick for a while. Some punch outs may be helpful. This can be solved with setting some dead band and is often an problem with PPM and PWM receivers. 

In general I would not see an reason why G-Tune should not work in 3D mode.

Another problem could be with the LUX PID controller the resolution of the tuning steps is likely not fine enough. Also the actual code does not allow you to go below 10 fro the P value (1.0 in the configurator).

Pomaroli, 2016-04-08:
thanks guys, i'm using PPM therefore might be the right direction.
Actually my deadband settings are

# get deadband
3d_deadband_low = 1470
3d_deadband_high = 1570
3d_deadband_throttle = 50
alt_hold_deadband = 40
deadband = 0
yaw_deadband = 0
accxy_deadband = 40
accz_deadband = 40

Shall i try to set deadband = 8 ?

Pomaroli, 2016-04-08:
i've just tryed with deadband=8, yaw_deadband=8 did not work.
It's not easy with this beast, it's somewhat of fast but i've done also some hard punches without touching Roll and PItch in Manual Mode. Nothing changed.

MJ666, 2016-04-08:
you can see how much jitter you have in you signal in the configurator receiver page and set dead band accordingly. What mean yaw dead band will not work? are you using LUX pid controller?

I personally don't use G-Tune anymore since a while. With black box and the new filters is getting more easy and accurate to tune manually. Using EZGUI on my phone and an USB host cable to connect so I can tune easily in the field.

Pomaroli, 2016-04-08:
Sorry i change deadband and yaw_deadband to 8 and this did not take any effect on gtune.
I'm using the default Multiwii (rewrite) PID Controller.
There is no significant "jitter" in the signal...very stable

Pomaroli, 2016-04-08:
When configured as 3D, cleanflight will ask for an arm switch, could this be the reason ?
Could it be possible to change LED Color or special beep tone to singalize that the copter is in gtune mode, that would be cool ?

MJ666, 2016-04-09:
I'm using an switch to arm all my copters. This was never a problem with G-Tune. Unfortunately I don't have na 3D capable copter and likely also not the skills for 3D flying (yet). With the LED it cold be an enhancement for the future. G-Tune itself could be helpful to find some starting point but not a perfect solution yet. The algorithm also requires to had I and D in an good range already. In most cases the default settings fro this work quite well. Likely manual tuning is still always required.

Pomaroli, 2016-04-10:
So i can only get you there if gtune will be discontinued it's up to you.
But Anyway thanks for hint, so today i tuned the Bad Boy manually in flight. 
The value enden up to 
Roll/Pitch
P 8.4
I 0.45
D 80 
D seems really high is this plausibel ?

DasSmeagle, 2016-04-19:
Hi, 

I'm trying GTUNE on a standard quad and I too see no raction after activating GTUNE Mode in fglight. nochange in PID after landing, no noticeable raction in flight... Is this still working? Using 1.12.1 on a Naze32 compatible. Westerlund 32 if that is important.

I don't have big noise (all sticks at 1500 +/- 2) , but set Deadband 10 just to be sure. I checked the PIDs without removing battery after flight.

Thanks,
- Oliver

MJ666, 2016-04-19:
In general as you describe it should work. There is some more information in the comments above. Be sure you don't do any Roll and Pitch input for a while. As long this sticks are moving the tuning will not start.

MJ666, 2016-04-24:
Looks to be this is the problem why tuning does not happen anymore? The issue is likely related to some recent changes: https://github.com/cleanflight/cleanflight/issues/2074

DasSmeagle, 2016-04-24:
Yes, that sounds like it. I really have tried EVERYTHING I could come up with. Twice. An never got any change in PIDs with GTUNE...

hydra, 2016-05-06:
fix merged.

closed 2016-05-06 by hydra