cleanflight/cleanflight #2022: SPRacingF3EVO SPI mag driver causes stuttering motors.

hydra, 2016-04-07:
Until this is fixes disable the mag on the SPRacingF3EVO using the following cli commands

```
set mag_hardware=1
save
```

The issue can be seen in the GUI too - the looptime will occaisionally jump to around `24000` 10 times a second (the frequency of the updateCompass task).

The issue is the delay in reading from the I2C slave in the compass_ak8963.c file in the `ak8963SPIRead` method.

Possible solution: queue the I2C read in advance of needing the data.

MJ666, 2016-04-07:
We have seen this problem with the AlienFlight prototypes. The comunication to I2C via SPI is not ideal and will give even add more delay than native I2C. I was not able to reduce this delay much and on the AlienFlight F4 we have even more problems and comunicatin is not workig reliable yet. We have decided to go for the MPU6500 for newer versions and use an seperate mag sensor if needed in the future. For the AlienFlight F3 the mag is disabled by default already.

hydra, 2016-04-08:
i have a fix for this, working code here.  needs cleanup.  basically it adds a state machine into the mag read code and queues a read up, waits for the time to have elapsed and then reads the i2c buffer from the mpu9250.

i found it was possible to reduce it to two reads, one for the status, and another for the data and the status2 register.

hydra, 2016-04-08:
the general idea is this:

```
bool ak8963Read(int16_t *magData)
{
    bool ack;
    uint8_t buf[7];

    static ak8963ReadState_e state = CHECK_STATUS;

    switch (state) {
        case CHECK_STATUS:
            ak8963SPIStartRead(AK8963_MAG_I2C_ADDRESS, AK8963_MAG_REG_STATUS1, 1, &buf[0]);
            state++;
            return false;

        case WAITING_FOR_STATUS: {
            uint32_t timeRemaining = ak8963SPIQueuedReadTimeRemaining();
            if (timeRemaining) {
                return false;
            }

            ack = ak8963SPICompleteRead();

            uint8_t status = buf[0];

            if (!ack || (status & STATUS1_DATA_READY) == 0) {
                return false;
            }

            // read the 6 bytes of data and the status2 register
            ak8963SPIStartRead(AK8963_MAG_I2C_ADDRESS, AK8963_MAG_REG_HXL, 7, &buf[0]);

            state++;

            return false;
        }

        case WAITING_FOR_DATA: {
            uint32_t timeRemaining = ak8963SPIQueuedReadTimeRemaining();
            if (timeRemaining) {
                return false;
            }

            ack = ak8963SPICompleteRead();
/*
            uint8_t status2 = buf[6];
            if (!ack || (status2 & STATUS2_DATA_ERROR) || (status2 & STATUS2_MAG_SENSOR_OVERFLOW)) {
                return false;
            }
*/
            magData[X] = -(int16_t)(buf[1] << 8 | buf[0]) * magGain[X];
            magData[Y] = -(int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
            magData[Z] = -(int16_t)(buf[5] << 8 | buf[4]) * magGain[Z];

            state = CHECK_STATUS;

            return ak8963config.write(AK8963_MAG_I2C_ADDRESS, AK8963_MAG_REG_CNTL, CNTL_MODE_ONCE); // start reading again
        }
    }

    return false;
}
```

hydra, 2016-04-08:
there still needs to be some integration with the scheduler, so the updateCompass task can ask the driver if action needs to be taken or not.

borisbstyle, 2016-04-08:
@hydra 
Why is spracingf3evo missing exti definition in initialisation.c?

ledvinap, 2016-04-08:
@hydra: is the branch with rest of code anywhere? 

Why not use MPU slave handling to do all the work - just read registers containing data and wait for DRDY?

hydra, 2016-04-08:
@ledvinap i'm still looking into it at the moment, but thanks for the hints.  no the code isn't anywhere but on my laptop.  I'm waiting for my 4x4 to be fixed in the garage, limited internet access...

hydra, 2016-04-08:
@borisbstyle not sure what happened to that.  perhaps it was lost in the merge to master.

hydra, 2016-04-08:
I've pushed 4706dea to master.  It fixes the problem and doesn't affect other targets right now.  a properly solution and cleanup still required.

hydra, 2016-04-08:
improvement PR's welcome, i have to attend to other areas of code right now.

hydra, 2016-04-08:
closing this issue. I will open a new issue for the cleanup that is needed.

closed 2016-04-08 by hydra