cleanflight/cleanflight #2023: Error codes description needed.

maybeyes, 2016-04-08:
i have ran into a problem with sp3 board. It won't start and flashes led with 4 long blinks. Where I can find a description for error codes? It was mentioned in 1.10 release, but searching through whole manual brings no luck.

wkhard, 2016-04-08:
same here, searched the web without success. Can flash FW by bridging the boot pins, but not connect the SP board afterwards.
A hint would be mostly appreciated.

tks Kurt

maybeyes, 2016-04-08:
Yes, i also flashed FW with shorting the boot pins. Straight way flashing through cleanflight doesn't work and stops with "couldn't connect to bootloader" error. I tried several previous version of cleanflight - still no luck, it only changes the LEDs blink behavior. Then I removed the gyro ic, after that I got 2 long blinks. IMHO, this is not a gyro problem.

wkhard, 2016-04-11:
Tried up and down, in and out, without success.
Any hints from somebody?

MJ666, 2016-04-13:
Usualy the hard errors will be a series of long and short blinks (beeps if buzzer is also connected). Error codes are defined here: https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/system.h#L46
if you add one to each of this codes you get the number of long blinks (beeps).
So from you initial description it looks to be the ACC part of your MPU chip is not reacting as expected. Support for ACC revision 3 and 7 was added to the code recently. Did this board ever have worked before may be with an different firmware? Please be sure your are testing with the latest firmware. Hope you are also able to solder the MPU back or even can replace with an new one?
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu.c#L169
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu.c#L175
 With removing the MPU chip you get 2 long flashes wich is expected when the MPU is not present or not working at all for some reason.
https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L488

huste511, 2016-07-29:
I got an SP F3 10 DOF from Banggood this week.
It came with Cleanflight 1.12.0 and I can flash it to 1.13.0
I was able to connect with the cleanflight GUI and the boad works.

After that, I connet one BL-Heli-S ESC and the Suite are able to connect and flash the ESC FW.
Later i try to connect 4 ESC at one time. Im 100% sure with the wiring!
But the BLHelisuite claims to not able to connect anymore.

**Cleanfligt GUI are also not able to connet! ;-(
I got one blue LED on and a red LED flashes fast with 4 long blinks.**
That means FAILURE_ACC_INCOMPATIBLE?

Is this board bricked? With shorted boot pins I'm able to flash and erase 1.12 oder 1.13 again. But I am not able to connect ore flash the board normaly.

Perhaps BL-Heli bricks the board? Or an ESD?
But if the ACC realy gone, why I am note able to connect?

ketsa, 2016-08-04:
I also have a problem and need the error codes. 
My Sp racing F3 does 4 short 2 long flashes.

Link provided by MJ666 2 messages earlier doesn't lead to codes definition.
Please provide an updated  link to the error codes.

grizewald, 2016-08-05:
Yes it does. The link points to this enumeration:

typedef enum {
    FAILURE_DEVELOPER = 0,
    FAILURE_MISSING_ACC,
    FAILURE_ACC_INIT,
    FAILURE_ACC_INCOMPATIBLE,
    FAILURE_INVALID_EEPROM_CONTENTS,
    FAILURE_FLASH_WRITE_FAILED,
    FAILURE_GYRO_INIT_FAILED
} failureMode_e;

Add one to the starting number (zero in this case) and your two flashes mean FAILURE_MISSING_ACC. Cleanflight does not see the gyro chip, so unfortunately, your board is probably broken.

HomeY, 2016-08-11:
From the Betaflight Wiki (would be nice if it was in here also, also because the SPR manuals refers to the cleanflight Wiki:

**Why does my Flight Controller blink/beep lots of times when powering up ?**

5 short blink/beeps followed by any number of long blinks/beeps indicates an error code. Number of long blinks indicates the following error:

FAILURE_DEVELOPER: External interrupt of sensor failed to initialize.
FAILURE_MISSING_ACC: Accelerometer/gyro sensor is missing
FAILURE_ACC_INIT: Accelerometer/gyro sensor failed to initialize
FAILURE_ACC_INCOMPATIBLE: The found accelerometer/gyro sensor is not compatible/not the expected one
FAILURE_INVALID_EEPROM_CONTENTS: EEPROM/FLASH configuration content is invalid
FAILURE_FLASH_WRITE_FAILED: Write of configuration to EEPROM/FLASH failed
FAILURE_GYRO_INIT_FAILED: Gyro initialization of SPI MPU6000 accelerometer/gyro failed

closed 2017-02-25 by hydra

MarenB, 2017-05-27:
Hi,
please apologize, if this is the wrong place to ask. However, I ran into the same issue as described above. 
My SP Racing F3 stopped communication, after I set loop time and gyro sampling rate too high. I can flash all kind of software, but the CPU always interrupts startup procedure and keeps blinke 4x long. As I understand, this is MPU problem, I replaced it already twice with same type (MPU6050), but no change. W/o MPU it blinks 2x as expected.
Is it possible that there remains a code fragment or a setting in the EEprom which is not overwritten by "full erase"?
Or can te µC be fried? 
Any option to compile a debug-hex to see, what's going on?

Thanks,
Maren

TheStigFPV, 2017-06-22:
Ok, I'm having the same 4 fast flashes, 4 slow error flashes, then repeat on usb/battery startup. I am also unable to connect to betaflight. Everything was fine till I had a hard crash on my quad today. It would be nice if we all could find info or get clarification on this issue. I guess I broke my FC? This is the same SP Racing F3.

mikeller, 2017-06-22:
@TheStigFPV: See @grizewald's comment from Aug 5 2016. Your gyro has probably had it. :-(

TheStigFPV, 2017-06-22:
ok, thx.

mikief1, 2017-07-07:
Hi, I've been searching high and low for something like this. I'm getting continuous beeping during flight which when I continue flying turns from normal beeps to fast beeps. I can't see what's flashing. I thought it was too heavy or my camera was using too much power so I took it off. It is ok to hover around but if I try to turn it any faster than dead slow it is very unstable and often nose dives into the ground.
The beeps don't start if I'm hovering around my small garden until I put it under slightly greater movement. But its's barley a turn. Tried Angle and acro modes. 
I first thought the gyro might be fried but in acro it will stay nose down when i move it forwards - it's using the gyro to do that isn't it?
Any thoughts on what continuous beeping is? 
Sp F3 Evo brushless. It's this one:
https://www.unmannedtechshop.co.uk/sp-f3-evo-brushed-quadcopter-frame-with-fc-pdb-built-in/?gclid=Cj0KCQjw4vzKBRCtARIsAM3l8OCtAITsxbvkCGEmAmvCD2fphXvujusfVhxp1wSN399eu5QF7WRBy_8aAuikEALw_wcB 
(1st timer so apologies if I should start a new thread.)

mikeller, 2017-07-09:
@mikief1: GitHub is for discussions and questions around the development of cleanflight. For configuration questions, please go to RCGroups: https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please!!

Also, your beeping sounds like your battery warning levels are misconfigured.

Andre-Med, 2017-11-26:
I was having the same problem, (number of short flashes followed by 4 long flashes)

I fixed it by doing a firmware flash in clean flight with the latest firmware & the following settings
enabling 
- 'No reboot sequence'
- 'full chip erase'
- 'manual baud rate' (25600)

and disabling 
- 'Flash on connect'

hope this helps someone. Was worried I would have to wait for a new FC to arrive.

nikos-95, 2018-04-18:
I also had the same problem: SP F3 board from banggood, worked fine for half a year and then wouldn't start with 4 long error beeps. Tried to heat the gyro and its transistor (back side) with a lighter, no change. Now I put it in the oven at 140 °C, 15 minutes, and.. IT WORKS AGAIN!