cleanflight/cleanflight #2026: Cleanup MPU9250 SPI compass/mag support

hydra, 2016-04-08:
Currently the code includes temporary fixes to support the F3EVO board.  These temporary fixes need further cleanup and all the board specific code needs integrating properly for any board that uses the mag sensor on the MPU9250 via SPI (such as the new AlienWii boards).

See also #2022 for the initial bug report.

hydra, 2016-04-08:
FYI, See section 6.4.3.2. of the AK8963 manual for normal read operation.

http://www.akm.com/akm/en/file/datasheet/AK8963.pdf

blckmn, 2016-04-09:
@MJ666 might have some ideas here.

MJ666, 2016-04-09:
@hydra @blckmn will take a look into this again during the next days. The actual workaround looks to be an general solution but need some cleanup. The main issue is the delay needed for communicating with the I2C master inside the MPU9250. Not sure why Invensense is gone this strange route integrating the AK8936 into this sensor this way since it also supports SPI communication and there some pins available on the case to make this work. Anyhow nothing we can change. Finally make using the internal FIFO and make the driver interrupt and even DMA driven would be an solution.

@blckmn if you have some good ideas you can share. @hydra has made an commit for CF and BorisB has also merged this to BF already.

MJ666, 2016-04-14:
@hydra @blckmn I was working on the AK8936 driver last days and did some testing. Here is what i come up with. 

https://github.com/cleanflight/cleanflight/pull/1638/commits/1bc916756a7d211ca45e8ac45f82a34702bddf49
https://github.com/cleanflight/cleanflight/pull/1638

I did some more testing with different modes but this looks to be really the one wich least resource consuming. This code is bench testend on the AlienFlight F3 controller and the SPRacing F3 Mini. Likely it also need to be tested on the SPRacingF3 EVO but results should be similar than with the AF F3.

closed 2017-02-25 by hydra