cleanflight/cleanflight #2027: CC3D OPBL target is missing

phantom-8, 2016-04-09:
I understand CC3D openpilot bootloader option had been removed since v1.11 due to space issue.  The problem is that there are some CC3D boards that don't have Main & Flex ports (Eachine Racer 250 1st batch).  That means it's impossible to flash Cleanflight via main port, 'cos it's not present.  For these boards, the only way to flash Cleanflight is to flash the .BIN file via Openpilot GCS.

I've looked at the source code and the compilation option for OPBL had been removed.  Is it possible to add it back to the source code?  Users can disable other options, such as logging and make the image fit with OPBL.

martinbudden, 2016-04-18:
Yes, I'd like to see build support put back in. It would make my life much easier. CC3D is my goto development board and I spend too much time messing around with makefiles and linker scripts.

phantom-8, 2016-04-20:
I've manually put back the compilation option for OPBL and had successfully built a CC3D .bin image with GPS disabled.  But doing this for every release is tedious.  Can we have the OPBL target re-enabled in the Makefile, so the user can compile their own?

BTW, why is the GPS feature enabled for the CC3D target?  I don't think GPS is even possible on CC3D board.  If GPS is removed, we would have enough space even for a standard build.

Zero9r, 2016-04-20:
What do you mean GPS doesn't work with CC3D.  You are so wrong.  I have a GPS (UART1), Baro and Compass (Both parallel on I2C on UART2) installed on my 550 Whitesheep, plus about to add a USR 900mhz Telemetry Radio (Soft Serial).  All on a CC3D Running Cleanflight's iNav fork.  Even works on Normal CF.

OPBL is holding you all back.  Forget about using it.  Just get a FTDI Cable.  It cost me $3 from Banggood and took a week to get.  It just Plugs into the Flex Port (UART2)  and you can Flash HEX files right in Cleanflight.  plus it also get rid of the issues of having to use a battery to connect (if you use the FTDI not the onboard USB to connect) to the Configure and do not have to unplug and reset everytime you reboot during setting changes.  Makes more sense then fighting with OPBL and having it waste space.

closed 2017-02-25 by hydra