cleanflight/cleanflight #2030: Sane migration system

jevring, 2016-04-10:
Today each release is prefixed with "don't forget to take a backup" and "you have to do both a gui and a cli backup". While I understand that this is a hobby thing, it would be fantastic if we could at least get a single backup system, rather than two conflicting ones. Restoring one after the other has a tendency to leave some settings missing. For instance, I don't think I've ever managed to correctly restore aux settings. 

Better still is an actual migration system. You update to a new firmware, and the flasher reads all your old settings, transfers the ones it can, and asks about what to do for things like new "better" defaults. If the old value is equal to the old default, use the new default. If not, ask. 

You can only do this song and dance so many times before you simply settle on the version you have and stop updating. Can you imagine how many more testers we'd get for new versions if I knew that I could upgrade, and downgrade, at the drop of a hat without fear that my settings would be destroyed...

readerror67, 2016-04-10:
Sounds nice, where is the pull request?

jsweeney91, 2016-04-11:
Is this not a configurator change?

swilkens, 2016-04-11:
This sounds like something we should look at after we drop CLI: https://github.com/cleanflight/cleanflight/issues/922

jevring, 2016-04-11:
@jsweeney91 that's a good point, it almost certainly is.

jsweeney91, 2016-04-11:
@jevring might be worth posting it on there then

dkisselev, 2016-04-12:
Check out #1049 and all the more recent updates to it: #1976, etc.

There's a big effort in progress right now to migrate all configurations to a more manageable system.

Config parameters will live in specific groups and updates will explicitly nullify/clear/migrate just their own group, so upgrading and restoring backups would work much better.

closed 2017-02-25 by hydra