cleanflight/cleanflight #2031: CJMCU no longer works

martinbudden, 2016-04-11:
If I reprogram a CJMCU to the latest version of the software in the configurator, then I cannot get it to connect.

I'm pretty sure that the commit that broke things is this one:

https://github.com/cleanflight/cleanflight/commit/f54fa52c129076dd13d171decf02d5105a1a5cb0

> Remove PORT103V support that was added in the commit for UART4/5 support.
> The PORT103R also supports UART4/5 and we don't want to support another development target right now.
> Cleanup USART/UART inconsistencies in STMF1/F3 uart implementions.
> Cleanup remove some F1/F3 specific code from PWM mapping code.
> Relocate usart TX configuration code from system.c into serial_uart.c.

(I did a git bisect to track it down a while ago, but unfortunately lost the results, but I am 99% sure it is the commit above that broke things.)

Probably something to do with "Cleanup USART/UART inconsistencies in STMF1/F3 uart implementations."

Unfortunately I don't understand this part of the code enough to offer any more help.

hydra, 2016-04-13:
thanks for the report, i will investigate.

hydra, 2016-05-10:
yeah, it was c031e8d4aa7383ff5a933be06d6355507e6eb552

hydra, 2016-05-10:
fixed in 7cc578db12854af3764d6a8d73ff0423987c464d

closed 2016-05-10 by hydra

martinbudden, 2016-05-10:
Good to see this fixed. It looks like this could have affected other F1 targets as well.