cleanflight/cleanflight #2037: Incorrect setting of pidController for ALIENWII in config.c

martinbudden, 2016-04-12:
The PID controller is set using:
```
    pidProfile()->pidController = 3;
```
however 3 is not a valid number of the pid controller. This means, when set in `pid.c`, it will default to `PID_CONTROLLER_MWREWRITE`.

MJ666, 2016-04-12:
The PR for correcting this (and more updates) is already pending for a while. It is ready to be merged. I was still waiting for the IO update to use some better code for the updated LED driver but this also can be addressed later on.

https://github.com/cleanflight/cleanflight/pull/1638

martinbudden, 2016-04-13:
Great, I'll close this once the PR is merge in.

martinbudden, 2016-05-11:
Fixed in https://github.com/cleanflight/cleanflight/commit/f57836e271640a732bba9b551f3721441912c66d Merge branch 'MJ666-AlienFlightF3_V2'

closed 2016-05-11 by martinbudden