cleanflight/cleanflight #2042: Driver problem with Alienflight F3 board

Pedro147, 2016-04-14:
I am not very Github savvy so I apologise in advance if this an inappropriate question here but I do not know where else to ask. As described in the Alienflight documentation I installed the driver from the DfuSe demo package as detailed here [http://alienflight.com/firmware-flashing-f3/](url) and then replaced the ST Driver with the Win USB driver using Zadig. However when I go the device manager and check the driver details it shows that it is using an STMicroelecronics dr
[image]
iver whereas it should have been replaced with a WinUSB driver by Zadig I assume. The board connects to Cleanflight and can be flashed OK but I have been advised my @mj666 that this driver is the wrong one for the Alienflight F3 board. Consequentially I am just wondering what my best course of action is seeing as this is my first real experience with Cleanflight except for setting up one Scisky FC that went without a hitch
[image]

MJ666, 2016-04-14:
@Pedro147 as already stated in the RCG tread if you have flashed the board and can communicate with the CF Configurator to the board you are completly fine. Only the STM DFU boot loader driver need to be replaced. Since you already flashed the board it is already working as expected. You likely misunderstand my message in the RCG forum thread and you showed some other screen shoots there wich show problems with missing or improper installed drivers.

SteveAmor, 2016-04-15:
Hi @Pedro147 can you please close this issue. If you still need support, can you continue to use RCGroups or other forums. Many thanks.

Pedro147, 2016-04-15:
Sure @SteveAmor I am not well versed in Github protocol and so was going to leave it open seeing as my issue has not been resolved. Thanks

closed 2016-04-15 by Pedro147