cleanflight/cleanflight #2045: Include current PIDs in blackbox log

timdonovanuk, 2016-04-15:
It would be great to have a record of the current active PID settings stamped into the log, as you can then  clearly see what effect the changes made. After a long day of tuning we end up with dozens of logs and after dozens of PID changes it would be great to track the changes made.

Even better would be to dump the PIDs only on arm, and even better would be also respond to in-flight or stick calibration changes and record these too.

timdonovanuk, 2016-04-15:
And FWIW this looks quite simple to do, as the code already dumps rcRate and min/max throttle. I think it might be possible to just add another case line to blackboxWriteSysinfo(), similar to this:

case 5:
            blackboxPrintfHeaderLine("rcRate:%d", controlRateProfiles(getCurrentProfile())->rcRate8);

Arakon, 2016-04-15:
I agree, that would make things MUCH easier to track.

GaryKeeble, 2016-04-24:
This is something I've been doing on betaflight. I have it working for betaflight v2.6.2 but it is still in testing. I've loaded it to my quad and it works fine on the bench but I need to get out and fly it before putting through an official pull request. I don't see why it can't be put back into cleanflight too if enough people want it. I have a thread running on rcgroups if your interested [http://www.rcgroups.com/forums/showthread.php?t=2649495](url)

csurf, 2016-04-27:
upvote for this. Seems like a no-brainer. I just barely started working with BB logging, and was surprised to find that the PID constants weren't being logged. Makes the logs useless if you've accumulated many logs during a tuning session wherein you've made a lot of PID adjustments.

what's the status on getting this pushed into CF?

GaryKeeble, 2016-04-29:
As my changes are being incorporated into Betaflight's logging, will they automatically find their way back into cleanflight? I know there are some variables now being logged that are only available in betaflight at the moment so there would be some **very minor** changes required to make it work on cleanflight.

martinbudden, 2016-04-29:
@GaryKeeble , things don't "automatically" find there way back into Cleanflight - someone needs to create a PR. This looks like something that @hydra would like in Cleanflight. I've had a look at your changes, and as you say there are some minor changes required to work on Cleanflight. Would you be able to create a PR for Cleanflight?

@hydra has a lot on his plate, so I doubt the PR will be addressed immediately, but it should eventually make its way back into Cleanflight.

GaryKeeble, 2016-04-29:
@martinbudden 
OK thanks for the heads-up; I'll fork a copy of cleanflight; implement the revision with the modifications and get a pull request going A.S.A.P.

GaryKeeble, 2016-04-29:
@martinbudden 

Help - I have tried to fork a copy of cleanflight to my github so that I can modify the blackbox for cleanflight but it just keeps jumping to my betaflight repository and won't create a fork.

Locally on my Mac,I have already generated the changes required, they are just 2 files, and it compiled fine in eclipse. So the code is done, waiting for someone to try it out.

Any ideas (about why I can't fork cleanflight)?

martinbudden, 2016-04-29:
@GaryKeeble , github won't allow you to make more than one fork of repository "tree". So if you have already forked betaflight then you cannot also fork cleanflight.

The good news is that you don't actually need to create another fork. Just clone cleanflight into a directory on your Mac. You can then manually set up a `git remote` in that cleanflight directory to point to your betaflight repository, and push your changes of your cleanflight code to a branch in your betaflight remote repository. Then go to your betaflight repository in github and switch to this new branch. Then click the "new pull request" button. You will get a red "can't automatically merge" message - at this point click the left hand "base fork" button (which will say something like "borisbstyle/betaflight") and choose the "cleanflight/cleanflight" base repository - the message should then go green and you can create the pull request.

The above directions are from memory, so they may not be exactly right, but hopefully you get the gist of the idea. It all seems a bit mad, but it works because all forks have equal status in git, that is if A is a fork of B which is a fork of C then A is a fork of C and C is a fork of B and B is a fork of A etc - no fork is considered the "base fork".

GaryKeeble, 2016-04-30:
@martinbudden 

Thanks for the help, I have done this and submitted a pull request with the modification.

closed 2017-02-25 by hydra