cleanflight/cleanflight #2047: Feature request - choose LED Color for "armed" and "disarmed"

ps915, 2016-04-15:
Would it be possible that the user can change the LED Color for "armed" and "disarmed" state in Cleanflight and Betaflight?
Right now, the disarmed Color is blue and everytime there is a new releases, i have to change it manuell an compile it again.

Its not nessesary to configure it via the GUI. two cli variables would be awesome.

Cheers,
Phil

ps915, 2016-04-15:
https://github.com/MaxWinterstein/cleanflight/releases/

A fried did this for some time, but it's anoying to do this on every Release or even when switching the an other Color.

GlobalhawkDE, 2016-04-16:
Definitely a must! +1

gaelj, 2016-04-16:
PR #2050 (only the last commit) will enable you to user-define the armed color with minimal code change (1 line). Disarmed color is still static green however.
I don't see an easy way to adapt the existing code into assigning 2 custom colors to a given mode.
On the other hand  it could be part of a broader project enabling customization of all static colors.

JohnOCFII, 2016-04-16:
It seems this request could become a support problem.  If people start changing their colors to whatever pleases them, and then there is a support issue, and someone trying to help them asks "is the LED red?" for example but the user has changed their LED scheme, it could lead to confusion.

NightHawk32, 2016-04-16:
I don't think that will happen, you can always access the configurator. There are hundreds of copters without LEDs and those people also managed to get it running.

ps915, 2016-04-16:
Where is the problem to get the hardcoded colors into cli variables. 
Everything will be as before but if you want, you can change the color via CLI?

hydra, 2016-04-16:
deffo think there's no support issue.  support issues come from the FIXED leds.  plus support staff could always ask users to reset to defaults and retry.

hydra, 2016-04-16:
@gaelj see review of PR #2050

ps915, 2016-04-16:
Hey Dominic,
thanks you and all for the smooth communication. So whats the way my feature request will be come along?
Do i need to deliver more information on this and will it be implemented in Cleanflight? And whats about Betaflight? 
Will it also implelemted in Betaflight?

Cheers,
Phil

gaelj, 2016-04-16:
Hi, it's currently going through the build server.
Please give me your feedaback on 
https://github.com/cleanflight/cleanflight/pull/2053/commits/d2dd5260975482b3b43aa14683535c533adc75f2

skaman82, 2016-04-16:
Hi Philipp, this issue tracker ist cleanflight only. And yes this feature request will almost likely make it into CF.

JohnOCFII, 2016-04-16:
> It seems this request could become a support problem. If people start changing their colors to whatever pleases them, and then there is a support issue, and someone trying to help them asks "is the LED red?" for example but the user has changed their LED scheme, it could lead to confusion.

My Bad.  Sorry.  Too early. Not enough coffee. I was thinking someone wanted to change the colors of the fixed LEDs.

gaelj, 2016-04-20:
https://github.com/cleanflight/cleanflight/pull/2053

skaman82, 2016-05-11:
Done, ps915 please close this issue

ps915, 2016-05-12:
Thanks guys. :)

closed 2016-05-12 by ps915