cleanflight/cleanflight #2049: nav_rth_altitude - units?

rholt51, 2016-04-16:
I'm very close to getting this thing working, but could you please point me in the right direction on where I should go to get the units that CLi functions like "nav_rth_altitude" uses? I was assuming Feet - but I think I saw something somewhere that it was in cm. big difference, I know. if there's somewhere that defines allthese units, please send me the link or something.

min_rth_distance ?

also, is "nav_max_speed" the setting that the craft will use in "rth" mode?

thanks in advance,
Russ from Coral Springs, Fl, USA

martinbudden, 2016-04-16:
Are you using Cleanflight or iNav? I don't think this parameter exists in Cleanflight.

As a general principle, all units are metric. To answer your specific question, for iNav, `nav_rth_altitude` is cm. See:

https://github.com/iNavFlight/inav/blob/master/src/main/config/config.c#L227-L236

This is a support question, not an issue, so in future could you ask similar questions in one of the forums.

Can you please close this issue.

rholt51, 2016-04-16:
Martin,

Thanks for getting back to me.

I am attempting to use iNav. I believe my quad was behaving as it was told, but I had the unit types wrong in my CLi commands.

Thank you SO much for the link. It clarified so much.

Also – I will use support from now on – but things on GitHub are confusing to me. Could you pls send me the link to the support page?

I will close this issue – very happy guy.

Russ from Coral Springs, FL, USA

 

https://github.com/iNavFlight/inav/blob/master/src/main/config/config.c#L227-L236

martinbudden, 2016-04-16:
@rholt51 I don't think there is a specific Cleanflight support page. People tend to ask questions in the RC Groups.

closed 2016-04-17 by hydra