cleanflight/cleanflight #2051: cc3d not reading receiver pwm outputs

gegtor, 2016-04-16:
When i move sticks nothing moves on screen

and only throttle channel show 885 rest 1500

here is my config:

Receiver mode : rx_parallel_pwm
Channel map: aetr1234
Rssi channel: Disabled
Cleanflight version: 1.12

gegtor, 2016-04-16:
And i tested my receiver with servo all channels are working fine

gegtor, 2016-04-16:
Solved i need to disable ws2812 leds :( any one know how have leds and pwm receiver at one?

Arakon, 2016-04-16:
The CC3D runs out of timers, chances are that your only option is a different FC.

gegtor, 2016-04-16:
Ok thx

gegtor, 2016-04-16:
is there a way to free up timers?

Arakon, 2016-04-16:
Not without removing other features.

gegtor, 2016-04-16:
but i have only leds turned on

Arakon, 2016-04-16:
I mean actually removing them from the code.

gegtor, 2016-04-16:
can someone remove them for me if its not big deal :3?

martinbudden, 2016-04-17:
@gegtor , Cleanflight does not support PWM receivers and WS2812 LEDs at the same time. There is no real way to do this.

I suggest you get a PPM receiver, they are reasonably cheap and it also much easier to wire.

Can you please close this issue?

gegtor, 2016-04-17:
Ok thx

gegtor, 2016-04-17:
Ok Thanks

closed 2016-04-17 by gegtor