cleanflight/cleanflight #2060: Integrating with Raspberry Pi for autonomous control

AdamCanady, 2016-04-17:
Hi all, I'm working with @lord on a research project at Carleton College combining computer vision techniques with probabilistic robotics to make an autonomous control system. 

Our current architecture has two components: 
- a Naze 32 flight controller that manages all the sensors (IMU, baro, GPS) and RX integrations to keep our aircraft level and responsive to input from the transmitter.
- a raspberry pi with multiple cameras

Our hope is that we can use the I2C bus on the Naze 32 to send data to the Raspberry Pi, then allowing for a small window where the Pi can interject a flight instruction directly into the `writeMotors` call that will allow it to make decisions about flight control.

Initially, we thought about adding a hook in the `TASK_GYROPID` task with a timeout (so the Pi could inject an instruction or the FC would timeout and execute the instruction it would have done originally — this is for fault-tolerance so we can ensure the aircraft will at least stay up), but to be more 'structurally-sound' with the project architecture, it may be a better idea to write our own task for this and include it in the scheduler with high priority. 

Which option would work best? Adding the hook into the `TASK_GYROPID` task or adding a new task? If we added a new task, how would the tasks share memory and how would we manage 'stale' commands?

martinbudden, 2016-04-18:
@AdamCanady , I think this is the wrong approach. Cleanflight already supports the concept of a configurator or a "ground station" which can change Cleanfight settings but also can control the motors. This ground station can be attached via usb but also via Bluetooth.

The Cleanflight configurator is the obvious example (you will note that you can control the motors from within the configurator), and EZ-GUI is another example. Here's EZ-GUI controlling a quadcopter:

http://ez-gui.com/manual/controlling-a-model-from-the-ez-gui/

This is an app running on Android communication with and controlling the quadcopter using the MSP protocol over a Bluetooth link.

My approach would be to do something similar on the Raspberry Pi - get the MSP protocol going, probably initially over a serial link, and then later over a Bluetooth link if you want.

I think hacking directly into `writeMotors` would be problematic.

martinbudden, 2016-05-06:
@hydra , there's been no further response from @AdamCanady , so can you close this issue?

closed 2016-05-30 by martinbudden