cleanflight/cleanflight #2063: How to use ADC in APM 2.7.1

sriramgbhagwat, 2016-04-19:
i want to use the ADC to read the voltage and also to display the voltage in mission planner for example i want to use a temperature sensor to give voltage as input and want to see that values in Mission planner can this be possible? Also can i add new ADC value in status bar? please guide me how to do this.....thank you.

Arakon, 2016-04-19:
Uh.. I think you're in the completely wrong git here.

martinbudden, 2016-04-23:
@sriramgbhagwat , can you close this issue?

sriramgbhagwat, 2016-05-05:
Can someone Please tell me How to use the ADC in APM ardupilot(which have ATMEL ATMEGA 2560 controller which is connected to the analog pins A0-A8) and please correct me if i asked something wrong@Arakon. cant we configure the ADC pins of APM 2.7.1. 
EX: In mission Planner i can configure the A13 pin to read voltage of the battery and see the same in the window. So can I use the same to read other voltage readings

digitalentity, 2016-05-05:
@sriramgbhagwat this is Cleanflight firmware and compatible flight controllers. APM issue tracker is here: https://github.com/ArduPilot/ardupilot/issues

closed 2016-05-05 by digitalentity