cleanflight/cleanflight #2064: Supporting Parallax 28015 Sonar

gaelj, 2016-04-20:
I have this sonar with a 3 pin connector, Parallax 2815 rev C.
From the documentation I can see it is currently not supported by Cleanflight, so I though it would be nice if I could write a driver.
Implementation seems pretty straightforward:
- common signal pin to be triggered by an output to generate a burst
- reconfigure CPU pin as input and count the time until the echo burst is received

I have a NAZE32 rev6 full board, so I was thinking I could use the FT pad (going to STM32 pin PA15).

Any comment would be appreciated.

MJ666, 2016-04-20:
As you describe it its alredy working very similar to the actual sonar driver. You should investigate if this already fits. If this is not the case you should extend the actual driver by reusing the pins which are used by the driver already. In the worst case only the timings need to be adjusted.

gaelj, 2016-04-20:
No code change is required after all for this model.
Just connect two 1k R's to the board's sonar RX/TX (pins 7 & 8 of the RX header in my case where I use PPM and no SW serial ports). Connect both free ends together, and wire to the sonar's signal pin. Works great.

closed 2016-04-20 by gaelj