cleanflight/cleanflight #2067: How you guys DEBUG ?

gpolic, 2016-04-20:
Lets suppose that you are making a few changes in the code and want to observe some variable values to understand what is happening.
I do not have a dev board, so I was thinking what are the options.

Is there any way to show values on the LCD screen? (I have one but not connected it yet)
Any way to send variable data in blackbox thread and save it on the SD or Flash ?
There is a feature called Debug that can display values on the configurator, but I have not figured how it works. I looked into the debug source files thought. Is there any instruction/guideline ?

I have a SPRF3 and an old CC3D (for spare)
If no other possibilities, what dev board would you suggest ?

martinbudden, 2016-04-20:
Well, I tend to do it by writing test code and debugging that. I find the code, compile, debug cycle is much quicker than when you have to reprogram hardware. Of course sometime you have to run the code on real hardware, but there is a huge amount you can do within a test harness.

dkisselev, 2016-04-20:
Someone with more experience can pitch in about using debug, but it looks as simple as just setting the values, take a look at how the scheduler uses the debug variables:

https://github.com/cleanflight/cleanflight/search?utf8=%E2%9C%93&q=SCHEDULER_DEBUG&type=Code

Uncomment the debug flag in `scheduler.h` and `scheduler.c` will just overwrite the debug variables that are defined in debug.c/debug.h

I might be wrong but I can't see anywhere you would need to explicitly enable sending the debug commands - they should just flow automatically as part of the MSP serial stream.

I agree with martin about writing test code first though. I do it in all my projects now, even the lazy ones that I'm not going to bother with high code quality on.

Set up your testing framework right away, and if you ever need a clean slate to instantiate a class/function you're working on and see how it behaves, just throw some stuff into a unit test and fiddle with the function until the test passes.

nathantsoi, 2016-04-20:
st-link + openocd: http://nathan.vertile.com/blog/2015/10/24/st-link-flash-and-debug-on-the-spracingf3-discovery/

SteveAmor, 2016-04-20:
Yes, you just set the debug variable [0 to 3] to the value you want. You need to ensure debug.h is included. It automatically gets sent to the serial stream for viewing in the Configurator.
But I agree a unit test is best.

gpolic, 2016-04-20:
I must admit I am so old that when I was learning, this unit testing stuff didnt exist. Now I am just trying to catch up.  Thought that the test area was just to find conflicts or some errors that wouldnt compile anyway. Have a lot of studying to do :-o

At first I will try to use the debug variables, as it seems very simple for now
so include debug.h and go from there :-)
Thanks for taking the time to share this info guys

closed 2016-04-20 by gpolic