cleanflight/cleanflight #2071: Plane horizon roll reverse

wookie666, 2016-04-22:
I have everything else working with cleanflight and an Airplane.
The issue I am having is, i had to reverse my aieleron channel in my transmitter because with stock settings, when rolling to the right the plane would roll to the left.

Reversing it in the transmitter fixes the issue in passthrough. But the only issue is when using acro or horizon modes. the ROLL axis is trying to roll the plane, either left or right. So when it tries to roll right to relevel. it actually rolls left. I do not have the ability to reverse the servo position on the plane to allow this as theyre fixed in already.

Is there a way to specifically reverse the signal being sent out. So that it sends a left signal when trying to roll right and a right signal when trying to roll left.

if i adjust my board alignment. I can get the roll axis to work correctly, but then my pitch or yaw axis does not work properly

Thanks.

wookie666, 2016-04-22:
actually think I was able to acheive this using -100 as a direction and rate in the servo tab.

martinbudden, 2016-04-23:
@wookie666 , can you close this issue?

wookie666, 2016-04-23:
will do. Sorry about that

closed 2016-04-23 by wookie666