cleanflight/cleanflight #2072: [rx.h] Possible mistake in RX_FAILSAFE_MODE_COUNT value ?

scrouillou, 2016-04-22:
Browsing through the code, I came across those lines

{ src/rx/rx.h, lines 85-99 }

```C
85. typedef enum {
86.     RX_FAILSAFE_MODE_AUTO = 0,
87.     RX_FAILSAFE_MODE_HOLD,
88.     RX_FAILSAFE_MODE_SET,
89.     RX_FAILSAFE_MODE_INVALID,
90. } rxFailsafeChannelMode_e;
91. 
92. #define RX_FAILSAFE_MODE_COUNT 3
93. 
94. typedef enum {
95.     RX_FAILSAFE_TYPE_FLIGHT = 0,
96.     RX_FAILSAFE_TYPE_AUX,
97. } rxFailsafeChannelType_e;
98. 
99. #define RX_FAILSAFE_TYPE_COUNT 2
```

Ain't the value for `RX_FAILSAFE_MODE_COUNT` (possibly) invalid ?
Ain't the magic number for `RX_FAILSAFE_TYPE_COUNT` a source for future pain ?

Being a C++ developer, my C is pretty rusty and I could be mistaking, but wouldn't it be better — like it seems for the vast majority of other enums — to redefine those as :

`#define RX_FAILSAFE_MODE_COUNT (RX_FAILSAFE_MODE_INVALID + 1)`
`#define RX_FAILSAFE_TYPE_COUNT (RX_FAILSAFE_TYPE_AUX + 1)`

martinbudden, 2016-04-23:
Well spotted!

digitalentity, 2016-04-23:
Actually this is not a mistake. RX_FAILSAFE_MODE_COUNT defined number of possible failsafe modes for an r/c channel (used only in CLI)
For flight channels (R/P/Y/T) this is AUTO, HOLD, INVALID
For AUX channels: INVALID, HOLD, SET

scrouillou, 2016-04-24:
Thanks for the explaination @digitalentity 

May I suggest a comment based on your answer be integrated into the code in order to clarify the situation for future reviewers ?
Furthermore, based on your reply, it seems to me it would be a bit clearer to switch those two enums declaration, giving the following code :

{ src/rx/rx.h, starting at line 85 }
```
typedef enum {
    RX_FAILSAFE_TYPE_FLIGHT = 0,
    RX_FAILSAFE_TYPE_AUX,
} rxFailsafeChannelType_e;

#define RX_FAILSAFE_TYPE_COUNT 2

// possible failsafe modes for a channel
// For RX_FAILSAFE_TYPE_FLIGHT: AUTO, HOLD, INVALID
// For RX_FAILSAFE_TYPE_AUX: INVALID, HOLD, SET
typedef enum {
    RX_FAILSAFE_MODE_AUTO = 0, // only for RX_FAILSAFE_TYPE_FLIGHT
    RX_FAILSAFE_MODE_HOLD,
    RX_FAILSAFE_MODE_SET,      // only for RX_FAILSAFE_TYPE_AUX
    RX_FAILSAFE_MODE_INVALID,
} rxFailsafeChannelMode_e;

#define RX_FAILSAFE_MODE_COUNT 3 // only 3 values possible for each rxFailsafeChannelType_e, see above
```

I can put it in a PR if it is more appropriate, but i do not want to pollute the requests for such a cosmetic change, so if one of you is currently working around this piece of code and want to integrate it, be my guest

closed 2017-02-25 by hydra