cleanflight/cleanflight #2074: updateGtuneState no longer called in taskMainPidLoop

martinbudden, 2016-04-23:
At some point the call to `updateGtuneState()` in `taskMainPidLoop()` in `mw.c` seems to have been lost.

The function `updateGtuneState` is still there, but it is no longer called.

SteveAmor, 2016-04-24:
Git bisect is telling me it was removed here:  cb941a7041a7ddcb70e6a3f1859399dd25a6c5b8

martinbudden, 2016-04-25:
@SteveAmor , thanks for doing the git bisect. It looks like this was a genuine mistake when the new scheduler was incorporated. @digitalentity , would you care to comment?

SteveAmor, 2016-04-25:
Some of the PRs have waaaaaaassy tooooooo much going on.............

digitalentity, 2016-04-25:
Martin, this is high probability. I probably missed gtune in the first
place.

SteveAmor, 2016-04-25:
Guys, you are doing fantastic work :-)

hydra, 2016-05-06:
fix merged.

closed 2016-05-06 by hydra