cleanflight/cleanflight #2079: Angle Mode Drift - Not visible in sensor view

DasSmeagle, 2016-04-24:
Hi,

I have a problem in "Angle" mode... I have to hold the Stick to some position in flight to get the Copter to level. The exact spot changes with every flight.

I can see this behavior on the ground, too. Did a little video to show this:

https://youtu.be/D4UP3qolMZg

After I move the copter a little, the "artifical horizon" in OSD will be off to some direction. When I put it down/level, it will go back to "neutral" in a few seconds.

But in fflight, it will not go to neutral, I will have to hold my stick in a position "forever" to keep the copter stable at that point in angle mode.

It's not the sensor, when I connect with cleanflight I cannot see any drift there.

Can this be some setup problem or is something broken?

(Video is still uploading, should be there in a few minutes)

Thanks,
- Oliver

DasSmeagle, 2016-04-24:
Hi again,

did another Video. This time in flight.

You will see: On the ground its level.

In flight the "Artifical Horizon" goes up (but the copter is still more or less level on one point). After landing the AH slowly drifts back to neutral position.

https://youtu.be/IoMRUuZ81aM

Thanks for any input,
- Oliver

DasSmeagle, 2016-04-25:
Hi,

is there a better place to ask this? 

Thanks,
- Oliver

joshuabardwell, 2016-04-25:
This is probably being treated more as a support question than as a bug report or feature request. Consider asking in the Cleanflight thread over on RCGroups.

http://www.rcgroups.com/forums/showthread.php?t=2249574

However, I have heard many reports of this here and there, so it does seem that autolevel may be broken in the latest CF 1.12, at least under some conditions. Perhaps the issue should be taken more seriously by devs.

JohnOCFII, 2016-04-25:
This is probably a duplicate of #1780 

It appears the developers are considering a bug fix release to address this issue.

DasSmeagle, 2016-04-25:
Thanks,

I wasn't sure if I did something wrong or it is a bug... That's why I made the video and hoped someone with more experience can tell if bug or maybe setup problem...

I read that bug, but my problem is different, as I don't see any drift on the sensor data in cleanflight app. But at the same time I see the "Artifical Horizon" of OSD is clearly not centered and slowly moves to the center.

I don't know how this artifical horizon works, but shouldn't it be the same as sensor data?

- Oliver

martinbudden, 2016-04-25:
Yes, it seems this might be a duplicate of https://github.com/cleanflight/cleanflight/issues/1780

@hydra is there any chance of getting out a v1.12.2 with just this fix in? This issue keeps coming up again and again. PR https://github.com/cleanflight/cleanflight/pull/2076 fixes it. (You might also want to include PR https://github.com/cleanflight/cleanflight/pull/2081 which is a 3 line GTUNE fix.)

DasSmeagle, 2016-04-25:
If you need more info, videos etc. I would be happy to help...

mmormota, 2016-05-06:
I experienced the drift problem too.
160mm quad, Afromini, v1.12, multiwii rewrite, 1306 motors, blheli oneshot125 escs.
In angle mode the quad first flies level, then starts to drift back and right, the angle increases continously over time, after several minutes I had to push the stick half way to keep it level.
The problem somehow vibration related.
Using sorbotan gel in order attenuate vibration the drift improves, but still there.

**Important:**  using v1.10 on the same quad, no need for sorbotan, no drift at all.
Most probably something went wrong in the new version in acceleration sensor filtering.

Another problem, maybe related but not sure:
450 quad, Flip32+, v1.12, multiwii rewrite
Very hard to keep altitude, jittery. (angle mode, no altitude hold) Can't get calm flight with many PID mod efforts.
Same quad, v1.10, multiwii rewrite
Dead calm, perfect flight with default settings. 
Reproducible, others are trying as well, so pretty sure it's not my mistake.

hydra, 2016-05-07:
should be fixed in master now, see #2076

closed 2016-05-07 by hydra