cleanflight/cleanflight #2080: configuring acro naze32

helveticaneue77, 2016-04-25:
I'm kind of a noob at this so be gentle...

I am trying to build a drone from scratch & learning electronics at the same time...

I have an acro naze32 and am trying to set it up. 

plugged it into my pc using the microusb and using baseflight to configure. the app finds and connects to it, however, only gyro and accel show up as connections, mag and baro are red/not connected. 
(I've watched several tutorials on youtube and mag & baro always show up when the board connects)

any idea why? 

note: I soldered the pins on the board myself - is something not connected properly?

thanks for your help!!!

Arakon, 2016-04-25:
Wrong section for support questions, check rcgroups.
However, as a quick info: The whole point of the "acro" naze is to be for raw acro flying. It doesn't have a mag or baro.

SteveAmor, 2016-04-25:
@helveticaneue77 can you please close this issue.

closed 2016-04-25 by helveticaneue77