cleanflight/cleanflight #2082: Emax skyline32 v2 advanced problem

Jama85, 2016-04-25:
Hello

I updated from 1.10 to 1.12.1.
Now when i power up the esc not beeping. i can arm the copter but nothing happening.
When it powered up, and click to save and reboot on gui, the esc start beeping, and motors spin up.

Arakon, 2016-04-25:
Make sure your ESCs have a BLHeli version of 14.1 or higher on them, if they run BLHeli.

Jama85, 2016-04-25:
Hello

The esc is emax 12a nano with blheli, i will check.

Jama85, 2016-04-25:
Ok its just a 13.1

fape, 2016-04-26:
Oneshot is not working while gyro_sync is on (in 1.21.1 gyro_sync is on by default) http://www.rcgroups.com/forums/showthread.php?t=2443194&page=190#post34591917

Jama85, 2016-04-27:
Ok
i updated the latest firmware on the esc, one from four is not spin up, just stuttering :(

SteveAmor, 2016-04-27:
Hi @Jama85 please can you resolve this support issue in rcgroups (or other such forums) and close this issue so that developers time is not diverted from fixing Cleanflight bugs. Thanks.

Jama85, 2016-04-29:
Problem solved, bad soldering.

closed 2016-04-29 by Jama85