cleanflight/cleanflight #2086: SP Racing F3 + OLED = CPU freeze

TheBigFudge, 2016-04-26:
Hey guys I hope you can help me, I have a little problem.

I bought some of the small OLED from banggood and connected it via I2C to the FC to show some sensor data. It's working  but after some time the board kind of freezes and the red status LED is blinking. This happens when I power over Lipo and also over USB. The time when it "freezes" is changing, so sometimes it works several minutes and sometimes only a couple of sec. 
When it freezes the Cleanflight CLI  says the CPU is at 100% and the I2C error counter is at several thousand Errors. When I unplug and repower the board everything is working until I freezes again.

Somebody know how to help?  Haven't tried to fly because i'm a little bit afraid it will crash when it freezes.

proggod, 2016-04-26:
I think you need pull up resistors between the screen and your controller, google i2c arduino pull resistor oled

TheBigFudge, 2016-04-26:
Thanks for your help. Is this not only necessary when the OLED is not working at all? How does that effect the I2C errors?

proggod, 2016-04-26:
Well I am not certain, but was reading about the OLED display this morning and came across that.  From what I understand it prevents it from pulling to much power, and locking up the controller.    Try the recommended resistors and see if that fixes your issues.

TheBigFudge, 2016-04-26:
Okay I will try. Have you detailed information about which resistors and which pins? I found several info about that and always a different...

proggod, 2016-04-26:
I don't run a screen so I couldn't tell you exactly, but I am guessing if you just use the same one someone else does it will probably work, let us know your results.

TheBigFudge, 2016-04-26:
Okay I will have to wait some days until i'm back at my working bench, but thanks for the hint. Until then new advices are always welcome :D

proggod, 2016-04-26:
np

martinbudden, 2016-04-26:
This could be an I2C overclocking issue. Have you tried setting i2c highspeed off? That is type `set i2c_highspeed = OFF` in the CLI.

TheBigFudge, 2016-04-26:
I can't find that command. I currently flashed the lasted betaflight version. is this an issue?

martinbudden, 2016-04-26:
It looks like it is not available in betaflight. You might try flashing to Cleanflight and trying this out - at least that way you'll know if it is an overclocking problem.

TheBigFudge, 2016-04-26:
okay thanks a lot! I will test it.

proggod, 2016-04-26:
Keep us posted, I want to run an i2c oled eventually..

TheBigFudge, 2016-04-26:
So flashed back cleanflight and `set i2c_highspeed = OFF` --> no i2c errors and everything working perfect....no freezes. For testing purposes I `set i2c_highspeed` back to on and had since then a couple of freezes. So `set i2c_highspeed = OFF` solves the problem. Thanks a lot and @proggod give it a try....it's just a gimmick but look pretty cool in the dark ;-)

martinbudden, 2016-04-26:
Glad to be of help. Can you close the issue?

TheBigFudge, 2016-04-26:
Jap

closed 2016-04-26 by TheBigFudge